klausahrenberg/WThermostatBeca #179: ME81AH - choose internal/floor sensor

os62, 2021-01-18:
Would there a possibility to add internal vs floor vs both sensor selection on ME81AH?

Thank you.

klausahrenberg, 2021-01-18:
I will take a look into it

lozb36, 2021-01-21:
I receive lot of unknown mcu commands:
55 aa 03 07 00 05 2d 05 00 01 00
During analyzing mcu commands I wrote a note, these could be commands for the external floor sensor. Maybe this helps? 
I do not own a separate sensor, thats why I were not able to do dome further tests.

klausahrenberg, 2021-01-23:
The command is at the 6. place - your command is 2d - which is the byte for the floor temperature value at model ME81AH.
The selection of the sensor should be another MCU command. For my model, it seems the selection of sensor is byte 2b. For inside sensor selection it reports: 55 aa 03 07 00 05 2b 04 00 01 00. Do you see this command, when you select the sensor (internal, external, both) on the device?

klausahrenberg, 2021-01-23:
Ok, I implemented this for my device ME102H. I copied this implementation for your device ME81AH also with the command byte 2d. But I need a different byte for the temp value of the floor sensor - so it will not work with identical commands. So send me your other unknown MCU commands please.

lozb36, 2021-01-24:
You were right - the correct command is 2b - like your implementation for ME102H.

After changing "Advanced Settings" I receive the following commands:
55 aa 03 07 00 05 0a 01 00 01 01
55 aa 03 07 00 08 65 02 00 04 00 00 00 01
55 aa 03 07 00 08 1b 02 00 04 ff ff ff fe
55 aa 03 07 00 08 1a 02 00 04 00 00 00 0a
55 aa 03 07 00 08 13 02 00 04 00 00 00 28
55 aa 03 07 00 05 66 04 00 01 01
55 aa 03 07 00 05 **2b** 04 00 01 **00**

The last one with the command byte 2b changes:

00 -> Internal
01 -> Floor Sensor
02 -> Both

As I have not connected (because I do not own one) a external floor-sensor, my thermostat shows up "err" on the screen. I do not receive different commands after sensor selection exept for those above.
As I wrote before, maybe the 2d commands show the temperature from floor-sensor?

klausahrenberg, 2021-01-24:
Ok, here we go. Please test following in v1.20j: 
* Sensor selection should work now (byte 02b). 
* 0x2d is now floor sensor value again.
All of your other given unknown commands are ignored now. Please view, if other unknown commands appear.

lozb36, 2021-01-24:
Sensor selection via mqtt works.
But:
- I do not receive mqtt messages for sensor-selection changes (wheter changing sensor at device nor via mqtt). is "XXX/thermostat/properties/sensorSelection" the correct topic?
- I still receive unknown mcu commands {"error":"Unknown MCU command: 55 aa 03 07 00 05 2d 05 00 01 00"}

lozb36, 2021-01-25:
I Analyzed the other mcu commands send after "advanced Settings"

1.  Anti Frozen Function:
     55 aa 03 07 00 05 0a 01 00 01 01    
     On (01) / Off (02)
2.  Switching differential 1-5°C
     55 aa 03 07 00 08 65 02 00 04 00 00 00 01
     (01==1°C) (02==2°C)
3. Temperature calibration [-9°C to +9°C]
    55 aa 03 07 00 08 1b 02 00 04 ff ff ff fe  --> -2°C
    55 aa 03 07 00 08 1b 02 00 04 00 00 00 00 --> 0°C
4.  minimum Temperature
     55 aa 03 07 00 08 1a 02 00 04 00 00 00 0a    (0a==10°C) (0f==15°C)
5. maximum Temperature 
    55 aa 03 07 00 08 13 02 00 04 00 00 00 28     (28==40°C) (2d==45°C)
6. periods of time
    55 aa 03 07 00 05 66 04 00 01 01     -> 01: 5+2 / 02: 6+1 / 03 7+0 day mode

Maybe we can implement these information by time.

klausahrenberg, 2021-01-30:
Hi, I changed some things in v 1.20l:
 - Unknown MCU command will not arrive anymore. Function of ommand 0x2d is not clear, I have the same on my device ME102H, it will be just ignored now
 - Thanks for your advenaced settings analysis. I added this in comments in source code. I have no plan to implement this features. This commands are similar to my device ME102H - are you required to use one of these commands via MQTT?
 - The command byte for reporting of floor temperature is still unknown for me. Could it be that this is not a separate value at your device? That means, if you select 'internal sensor' the 'actual temperature' reports the temp value and if you switch to 'external sensor', the value of 'actual temperature' shows the value of the external sensor? Please check

lozb36, 2021-01-30:
Thanks for your work.
`are you required to use one of these commands via MQTT?`
No!

I am not sure. My thermostat shows err on the display after switching to floor or both sensors because I do not own a external sensor. I could not recognize any changes. I ordered a sensor at aliexpress to do some testing - but it will take some weeks for delivering.

chipmoca, 2021-03-24:
`I ordered a sensor at aliexpress to do some testing - but it will take some weeks for delivering.`

You can temporarily put a variable resistor of 10KΩ

ALERTua, 2021-03-29:
the seller has provided me with this
[image]

silvinojr, 2021-04-15:
> * The command byte for reporting of floor temperature is still unknown for me. Could it be that this is not a separate value at your device? That means, if you select 'internal sensor' the 'actual [...]

Hi klausahrenberg, many thanks and congratulations for the great work!

I have a ME81H  model with a floor sensor.
If I select "built-in sensor" on Advanced settings I get the built-in temperature on the temperature topic, as expected.
Selecting "floor sensor" I get the floor temperature on the temperature topic, not on the floorTemperature topic.
Selecting "both sensors" I keep only getting the temperature topic, always reporting only one temperature that looks much like the one coming from the built-in sensor. I have heated the sensors individually to get estimations.
I have also noticed that these are in line with what the unit's display is showing, I never got to see the Floor Temp icon on the display, always Room Temp.

Changing subject, something I've noticed is that the beta firmwares released after the modifications to deal with the issue #170 (state changes are not reported by MQTT immediately), I'm getting a somewhat unresponsive unit. In some cases it is reporting the actual temperature way to often. If the temperature falls on a value near a rounding point it sends mqtt messages every few seconds like 20, 20.5, 20, 20.5, 20, 20.5...
Maybe it would make sense to keep the reporting of actual temperature to a regular interval and only try to be close to real-time for the commands, targetTemperature, for example. Even for those, it seems that the ESP is struggling to formulate several mqtt messages in sequence. For the target temperature if you try to quickly increase/decrease several degrees, it makes a pause every 2 ou 3 button presses, a delay of a few seconds to report targetTemperature changes could solve that, aiming at sending only one message after the user stops inputting commands to the unit.

Besides that, I'm happy to make any tests that you need on the ME81H device.
All the best,
Silvino

ikke-t, 2022-08-30:
Hi, I also got such me81h device and learned to switch between internal/floor/both. I can verify it changes the reading device. However it never sends both the values in mqtt, just one temp.