klausahrenberg/WThermostatBeca #181: VH Control Calypso-W

a-dekker, 2021-01-20:
As my thermostat was using Tuya for remote control, I decided to give your firmware a try. No regrets so far!

My thermostat for controlling infrared heating is called "VH Control Calypso-W". There is not much info to find about the device itself. It looks like some rebranded device, sold in the Netherlands by a company called verwarminghandel.nl.

I can only find some Dutch info on their website: https://www.verwarminghandel.nl/calypso-digitale-inbouwthermostaat-wifi.html and in a PDF file: https://cdn.webshopapp.com/shops/226472/files/354139327/vh-control-calypso-december-2020.pdf

From the available thermostat models in your firmware, the ET-81W seems to fit best. On thing I noticed is that the floorTemperature and temperature values are swapped when selecting ET-81W. Setings targetTemperature and deviceOn work fine, I was not able to change the locked value.

I also noticed that the Webthings state report provides less info than mqtt; it does not provide locked and floorTemperature.

Any chance for "official" support for this device? If so, do you need some MCU command errors? As said, the most imported functions work fine, so I can deal with the current state.

klausahrenberg, 2021-01-23:
I implemented the support for your device based on ET-81W in 1.20g on. I swapped only actual and floor temperature. If you receive 'unknown MCU commands' via MQTT, send me the messages and the operation you maybe did at the thermostat.

a-dekker, 2021-01-23:
Thanks for your response. I am trying to flash 1.20g, but no luck so far. I flashed 1.19 initially using tuya-convert, but the firmware upgrade via the web interface only leads to a reset and no new firmware. The md5sum of WThermostat_1.20g_beta.bin file I am using is 2b633198a7b9a4ce4660154dad037f73.
Any suggestions?

klausahrenberg, 2021-01-23:
Not really, I switched today several times between 1.19 and 1.20 without problems with Firefox. You can try with Chrome. It shows the percentage of the upload. Sometimes the firmware can't be uploaded 100% - then the update fails, so it needs some more attempts.

a-dekker, 2021-01-23:
Already tried Firefox, Chrome and konqueror. Now I also tried Chrome and Edge from within a virtual machine and using Chrome from a Chromebook.

All fail with a "connection was reset / this site can't be reached" in about 12 seconds. The most promising sign is the Edge browsing that sometimes shows uploading 31% in its status bar before the error page.  I have flashed other ESP8266 devices in the past using OTA without issues.

a-dekker, 2024-08-06:
After all these years I now own a new modem-router due to switching provider, so I decided to give the reflash another go. And now the reflash of the latest firmware succeeded right after the first attempt.

But for some reason, everything reacted very sluggish on the display. restarts end up in AccessPoint mode most of the time and after a few minutes the device switches off.

I decided to go back to the release I wanted to try all along, 1.20g. FloorTemperature and temperature are no longer swapped when selecting the Calypso-W, so my issue looks to be fixed going from 1.19 to 1.20g. And I do not suffer the issues of the newest firware, so I will stick with this version and will close this issue.

closed 2024-08-06 by a-dekker