klausahrenberg/WThermostatBeca #188: ME81H pictures for documentation

Gulpman, 2021-01-28:
Hi, 
I ordered a ME81H - after some search on this page I found out that it should be the same as the ME81AH mentioned here. Searching the docs I fould pics from other thermostats but not from the ME81H.

I did some pics and hope they are helpful for the documentation.

[image]
[image]
[image]
[image]
[image]
[image]

I will do the soldering and flashing on the weekend. :) If I understood correctly, I must use V1.20 for this thermostat. 

Best 
Sascha

klausahrenberg, 2021-01-30:
Thanks for the pictures, I will add it to the docs. Yes, for this Thermostat use the latest firmware of 1.20. However, it seems that the support isn't fully working yet. I don't own this one, so every feedback is welcome to finish the implementation for this device.

closed 2021-01-30 by klausahrenberg

Elwell, 2021-05-17:
I've just ordered one of these and noticed the board rev has gone up to 1.2. Picture at https://twitter.com/elwell2000/status/1394228472541679617/photo/1

tsrubar, 2021-05-17:
> I've just ordered one of these and noticed the board rev has gone up to 1.2. Picture at https://twitter.com/elwell2000/status/1394228472541679617/photo/1

Indeed, i can confirm that. New design doesnt have underlight power button and print seems  cheap on these
[image]