klausahrenberg/WThermostatBeca #190: WIFI not visible after flashing firmware

Gulpman, 2021-01-30:
Hi, 
I do have a question, regarding the WIFI which is supposed to be there after flashing the thermostat successfully.

I'm using firmware 1.20k beta and ME81H thermostat. I have never configured / used the thermostat before, just hooked it up once to see if it is working and then directly went to the flashing instructions. I used a CH340 serial programmer. 

I think I successfully flashed the V1.20k beta firmware:

- did backup of original firmware
- erased flash
- and flashed your firmware (output following, MAC address Xed out):

```

e:\temp\esptool-master>esptool.py -p COM4 write_flash -fs 1MB 0x0 WThermostat_1.20k_beta.bin
esptool.py v3.1-dev
Serial port COM4
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: xx:xx:xx:xx:xx
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 411504 bytes to 289035...
Wrote 411504 bytes (289035 compressed) at 0x00000000 in 25.7 seconds (effective 128.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

e:\temp\esptool-master>
```

After that I removed every wire except 

- VCC to VCC 3.3V from the CH340 programmer
- GND to GND

and reconnected the CH340 to the USB of my computer.
I have soldered a bridge between GND and GPIO15 which is still there as I think this shouldn't be an issue if I got this article from ESPRESSIF correctly -> https://github.com/esp8266/esp8266-wiki/wiki/Boot-Process#esp-boot-modes 

I also did try to connect EN+VCC from the ESP to VCC of the programmer but with the same result: no WIFI visible starting with Thermostate_Beca 

As I think I did everything correctly (emphasize this on think ;) ), I was wondering if I missunderstood / forgot something or just did a mistake?

- Do I need to unsolder the bridge from GPIO15 to GND?
- Do I need to fully assemble the thermostate and plug it to 220V before the WIFI is visible?
  - If so: can I leave the bridge from GPIO15 to GND?
- Anything else as an advise?

Help would be appreciated.

Best
Sascha

lozb36, 2021-01-31:
Hey, you have to disconnect your USB adapter and connect the thermostat to 220V to get the wifi working. You have no wifi during flashing or when the esp is only connected to the flashing stick.  

I started the thermostat for testing with all soldered connections without any problems.

Gulpman, 2021-01-31:
Hi @lozb36,
thank you very much for your quick response. 
I did as proposed and .... magic ... the WIFI  was there and I could run through the configuration process. :) The thermostat is sending information packages to my MQTT server and the server seems to send packages back (just checked the log file - will go into deeper exploration tonight or tomorrow). 

Maybe it could be added to the documentation, that connection to 220V is necessary to proceed with configuration? I spent a while figuring out what the issue could have been and decided to ask for help before going through the tedious process of desoldering my connections.  

It could be something like 
"Be sure that your thermostat is fully assembled and connected to 220V before proceeding with configuration as the WIFI will not show up if your thermostat is only connected with a 3.3V connection when you have just finished flashing." at the beginning of https://github.com/klausahrenberg/WThermostatBeca/blob/master/Configuration.md

or (where I personally would prefer it)

"After you have successfully flashed the firmware to your thermostat you need to fully assemble it to proceed with its configuration. The web page for confuguration will not be shown if the thermostat is not connected to 220V. Do NOT run the thermostat with 220V while being not fully assembled!" at the end of https://github.com/klausahrenberg/WThermostatBeca/blob/master/Flashing.md

Thanks again for the great project and the fast help :)

Best
Sascha

klausahrenberg, 2021-02-01:
added in flashing instruction

closed 2021-02-01 by klausahrenberg