klausahrenberg/WThermostatBeca #191: Can't access Wifi mode after quiting flashing

icebrian, 2021-01-31:
Hello all, 

Have a strange situation. I was baiscally wanting to test just to see if I could flash via tuya-convert and so set-up everything, connected my mobile phone to vtrust-flash then places my BHT-002-GALW device into Wifi mode by turning off and holding right most button for about 10 seconds.

After entering wifi mode I continued flashing which went through  a backup and in the end I was presented with the menu to select firmware and continue flashing. At this point I Quit using the "q" letter.

The reason is I wanted to check logs make sure everything was ok.

Now however I can't get into Wifi mode on the device. I've disconnected from power various times and whatever amount of time I hold the right most button it never sends me to Wifi mode.

Anyone have any ideas?

Thanks!

lozb36, 2021-02-01:
Do you have some log Output from tuya-convert?

I guess it may erased the old firmware and waited to flash the new one. In this case, you would have all options from device (like temperature, locking) without any network-connections because these are from the device itself, the tuya-convert and this projects firmware flashes only the esp - the part which is responsible for the network connection.

If you find any logoutput like "erase flash" - you have to solder your thermostat and flash the device manually.

icebrian, 2021-02-01:
Hi thanks for your reply ozb36

Actually I do have all logs and console output.

This is the console output:

<pre>
tuya-convert v2.4.5
Checking for network interface wlan0... Found.
Checking UDP port 53... Occupied by dnsmasq with PID 501.
Port 53 is needed to resolve DNS queries
Do you wish to terminate dnsmasq? [y/N] y
Attempting to stop dnsmasq.service
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Occupied by mosquitto with PID 453.
Port 1883 is needed to run MQTT
Do you wish to terminate mosquitto? [y/N] y
Attempting to stop mosquitto.service
Checking TCP port 8886... Available.
======================================================
  Starting AP in a screen.
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
............................................................................................
IoT-device is online with ip 10.42.42.42
Fetching firmware backup
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 1024k  100 1024k    0     0  63615      0  0:00:16  0:00:16 --:--:-- 28926
curl: Saved to filename 'firmware-c01d98.bin'
======================================================
Getting Info from IoT-device
VTRUST-FLASH 1.5
(c) VTRUST GMBH https://www.vtrust.de/35c3/
READ FLASH: http://10.42.42.42/backup
ChipID: c01d98
MAC: 98:F4:AB:C0:1D:98
BootVersion: 7
BootMode: normal
FlashMode: 1M QIO @ 40MHz
FlashChipId: 1540c8
FlashChipRealSize: 2048K
Active Userspace: user2 0x81000
======================================================
Ready to flash third party firmware!

For your convenience, the following firmware images are already included in this repository:
  Tasmota v8.1.0.2 (wifiman)
  ESPurna 1.13.5 (base)

You can also provide your own image by placing it in the /files directory
Please ensure the firmware fits the device and includes the bootloader
MAXIMUM SIZE IS 512KB

Available options:
  0) return to stock
  1) flash espurna.bin
  2) flash tasmota.bin
  3) flash WThermostat_1.19.bin
  q) quit; do nothing
Please select 0-3: q
Leaving device as is...
======================================================
Do you want to flash another device? [y/N]

======================================================
Cleaning up...
Closing AP
Exiting...
</pre>

All other smarthack logs, mqtt, psk, udp, web and wifi that where copied into the "backup"dir do not have any "erase" word within them.

klausahrenberg, 2021-02-01:
I didn't used tuya-convert so far. But according to your log file: Could it be that tuya-convert already installed a 'intermediate firmware'. And this 'intermediate firmware' is asking, which third party firmware you like to install? So, to cancel the flashing, the right option was '0) return to stock' (this had flashed the backup again) and not option 'q'?

So did you tried to start tuya-convert again to resume from the last choice?

klausahrenberg, 2021-02-01:
The EZ config mode (blinking fast) is a result of interaction between MCU and ESP of the thermostat. So if you are not able to get the device in the pairing mode (blinking) again, then this would be a strong hint, that an 'intermediate firmware' is installed. But in this case you should be able to resume from the last options of tuya-convert.

icebrian, 2021-02-01:
Thanks @klausahrenberg that was indeed the case. I had actually tried it before, meaning setting tuya-convert to AP mode then turning off and holding down key but since I never saw anything in my smarthack-psk.log I thought the device wasn't connecting so never tried moving forward.

Today however I did try, and as before it started abackup then gave me the option, to which I choose "0) return to stock" and I can now once again access Wifi mode on the device.

Seems all is good and I can move forward with flashing! Thanks for your assistance and great work, indeed very much appreciated!

Best regards!

closed 2021-02-01 by icebrian