klausahrenberg/WThermostatBeca #194: Support for POER PTC20 (has humidity sensor)?

woowoosan, 2021-02-03:
Hi,

I did a search to find a thermostat with humidity sensor and the only device I found was PTC20 by POER: https://de.aliexpress.com/item/4000044931140.html

I first tried to find a way to integrate the thermostat into iobroker without modifying it, but that wasn't possible. So I flashed WThermostat_1.20l_beta.bin. With that FW the display is showing nothing and I think that the MCU is also not sending commands to the ESP. But the ESP itself is working, I can use the webinterface, configure network etc. So my conclusion is that this is not a Tuya-based device and cannot be supported easily.

So my two questions: 
Has anybody experienced a similar behavior? Maybe I'm doing something wrong?
Is anyone interested in this thermostat and may be able to help?

woowoosan, 2024-02-08:
No. I abandoned the Poer PTC20. I switched my focus to knx and there are devices that can measure both temp and humidity (although they are more expensive).  Sorry to disappoint you...

DerDom, 2024-02-08:
Interesting, what KNX devices are you using for humidity?

woowoosan, 2024-02-08:
The most cost effective device I know of is Lingg+Janke TA4F55TH-BCU-E (see https://lingg-janke.de/en/knx-eco-products/push-buttons-and-temperature-sensors-exclusiv-55/).

Other options with display: https://www.mdt.de/en/products/product-detail/sensors/assortment-55/push-button-smart-55.html or https://www.mdt.de/en/products/product-detail/sensors/controller-and-sensors/room-temperature-controller-smart-55.html (even with digital inputs!).

DerDom, 2024-02-09:
Thank you for the links. Are you connecting multiple devices via cable to a KNX/IP interface or router to access them? Unfortunately, I cannot add cables in my case.

I was considering taking OMS/868MHz sensors for measuring room temperature and humidity. There are many devices with a 10 year battery life (e.g. https://www.relay.de/produkte/m-bus-wireless/relair-ics). Using OMS would also allow me to collect data for humidity and temperature for all the heaters using wmbusmeters.