klausahrenberg/WThermostatBeca #195: add compatibility for TUYA TH213 (parameters attached)

Ruggero1912, 2021-02-03:
Hi,
I have a Tuya Thermostat for gas boiler, like the one in this pic:

[image]

I think that the model is Tuya TH213 but I am not sure, because I bougth it from a reseller and I did not found any reference about this model on tuya webiste. This is a link to the one I bought: [link](https://www.amazon.it/gp/product/B08CRSBGDP/). The reseller is AWOW.

I tried this firmware and all the thermostat configuration provided. But none of them works.
Here is a pic of the board:

[image]
I noticed that my thermostat is based on a TYWE1S instead of TYWE3S. This is the pinout for TYWE1S:

[image]

After I installed Tasmota and made some tests. I discovered that the GPIO pin for TuyaRX is GPIO13 and the one for TuyaTX is GPIO15.
I also used [the tool from sillyfrog](https://github.com/sillyfrog/Tasmota-Tuya-Helper) to determine what dpId does what.

[image]

This is what I understood:

> 1) boolean turn ON or OFF the thermostat
> 
> 2) integer target temperature
> 
> 3) integer current temperature
> 
> 4)switch mode enum 0 - 4
> 0 -> manual
> 1 -> home
> 2-> away
> 3 -> auto
> 4 -> ? set the temperature to 18°C, no icon shown. maybe eco or sleep mode
> 
> 6)boolean lock (lock the thermostat buttons)
> 
> 12)integer ?
> 
> 101)integer ?
> 
> 102)ENUM shown an error in thermostat display if set to 1 or 2, otherwise nothing happen, 0 default
> 
> 103)integer calibrates temperature's readings (only positive values allowed)
> 
> 104)?
> 
> 105) boolean ?

To contribute to the development I tried to build by myself a version of WThermostatBeca adding what I tought was necessary, but I did not find how to change the GPIO pins for Tuya RX and TX, moreover I found that the configuration is not easy at all.
Could you add the compatibility to this thermostat? Or tell me how to do to implement it?
Let me know if you need something more.

Thanks

Ruggero1912, 2021-02-04:
Meanwhile I continued trying to make it work, and it seems that I made something.
I've found how to set GPIO15 and GPIO 13 as RX and TX after a lot of trying: it is sufficient to call the method swap() on the object Serial, from the class HardwareSerial. You have to pass as argument of the method swap() the TX pin number, in my case 15.
[image]
I flashed the firmware, and now, selecting the new thermostat that I configured starting from the template, I got able to see the correct temperature. 😄 
I would like to help implementing the support of this thermostat, let me know what can I do next.

Pigotka, 2021-03-05:
Hello @Ruggero1912, I have same device. 
How did you flash Tasmota there? I don't see the GPI0 pin to put the device to flashing mode.

Pigotka, 2021-03-05:
Alright I managed to find all the pins. Did you use tuya-cenvert or did you solder?
Isn't tasmota enough to use this device or why do you want to use this firmware?

Ruggero1912, 2021-03-05:
Hi! Finally an answer! 😃
[image]
As in the first post, GPIO0 is the one at bottom left of the chip.

Ruggero1912, 2021-03-05:
I soldered the cables the first time.
Because of an error I had to flash it again, and that time I managed to
hold the contacts by myself without soldering - I know that it is not a
good way, don't try this at home ;) -.
How would you use it with Tasmota?
If you have an idea of a proper configuration let me know.
At the moment I am using this firmware with some changes made by me to
adequate the firmware to this hardware.

Pigotka, 2021-03-05:
So to my imagination I have tasmota in other devices in the house. After flashing and setting proper GPIO configuration it sends the data through MQTT.

You mentioned you have with working. What else do you need to make PR?

Ruggero1912, 2021-03-05:
How do you setup Tasmota to make it communicate with the MCU that handles the thermostat display?
This is the great advantage of this firmware.
To make the firmware work I have had to make some changes to the source code and compile an adequate version.

Pigotka, 2021-03-05:
Do you have the codes in some branch I can take a look? I can flash it over the weekend. If your changes are not final I can help you to finish it.

Ruggero1912, 2021-03-05:
Happy to read that!
I have the code on my pc, I will upload it tomorrow.

Pigotka, 2021-03-14:
Did you managed?

Ruggero1912, 2021-03-14:
Sorry for the delay. 
I think that I just managed to load the branch on github at this link:
https://github.com/Ruggero1912/WThermostatBeca
Not sure that this is the right way to publish a custom version.
In the case I will change things accordingly. 
As you can see from the commit, I called the thermostat "Paciosoft thermostat" as I am was not sure of the effective name.
I also edit the platform.io build configuration to let it know that our thermostat has a 2 MB ROM.

At the moment in the git there is no compiled version of my revision, so you should build by your own.
Later I will try to upload a build of this rev.

Let me know what you think, any suggestion would be really appreciated.

Pigotka, 2021-04-05:
Hey, just to dump this. I did not find the time to look yet. I'm a bit surprised that no-one else is doing it. THose thermostat are quite neat.

Pigotka, 2021-04-05:
Alright I actually got some time now to read your changes. You did good job. The code is still obviously WIP so it needs cleanup.
I assume "paciosoft" is your brand and should be changed to TH213. Beside cosmetics, you have some problem processing commands (afaikt). Can you tell us what is missing in the `processStatusCommand`?

After fixing this only problem that needs to be addressed is the increased RAM size to 2m. This would be issue for other thermostats which could have less. Would be nice to hear from @klausahrenberg how to proceed.

Ruggero1912, 2021-04-05:
Happy to read your reply.
Yes, the name paciosoft is a joke more than a brand, but at the time I wrote the code I did not managed to find the real name of the thermostat. not sure TH213 is the right code name.
If you compare the .h file that I wrote to the one of another supported thermostat you will note what is missing, at the moment I do not remember the list of what is not working. The main features are there, but the weekly planning that I see for the other thermostats is not implemented at the moment.
What I have incresead is not the RAM, is the ROM setup because the chip in our thermostat is a TYWE1S (based on a ESP8266EX) that has a 2 MB rom. If you leave the setup at 1MB the compiled version will still work both on our and on the other thermostats.
I did that change because it leaves me on the ROM enough free space to upload another firmware and make a direct flash via wireless from a version to another. Not necessary.

Pigotka, 2021-04-05:
Btw. do you know if those 3 pins works? I'm not fan of soldering directly to the TYWE1S. Maybe there are all the pins accessible on the board. Those three seems obvious but not sure if there is 3v and GPI0 on the board. I did not managed to flash it yet.
[image]

Ruggero1912, 2021-04-05:
I do not remember that. 
But I suggest you to use a multimeter to see if those are the same.. :)

Pigotka, 2021-04-12:
Alright, I managed to compile and flash thermostat and it works. Excellent job @Ruggero1912.
By pressing the button I can only switch between Home and Away modes. I discovered some inconsistency in some stats but nothing major/critical.
Now, if you don't mind cleanup the codes and prepare pull request.

Btw. do you know how to set temperature from outside using MQTT?

Ruggero1912, 2021-04-12:
Very good.
To control the thermostat from MQTT the procedure is the same as for the other thermostats:
- you have to set the right MQTT server in the configuration of the thermostat (using the web interface)
- you have to configure an MQTT sensor to get the read temperature like this schema:

> `#TERMOSTATO TUYA INTEGRAZIONE DEL 04/02/2021
> 
>   - platform: mqtt
>     name: "thermostat_current_temperature"
>     state_topic: "corridoiotermostato/thermostat/properties"
>     value_template: "{{ value_json['temperature'] | round(1) }}"
>   - platform: mqtt
>     name: "thermostat_target_temperature"
>     state_topic: "corridoiotermostato/thermostat/properties"
>     value_template: "{{ value_json['targetTemperature'] | round(1) }}"
> `
_change "corridoiotermostato" with whatever you set as MQTT client name, this configuration is for Home Assistant in the sensors list_

This is the climate entity configuration that I use:

> `  - platform: mqtt
>     name: termostato corridoio
>     unique_id: termostato_corridoio
>     temperature_command_topic: "corridoiotermostato/thermostat/set/targetTemperature"
>     temperature_state_topic: "corridoiotermostato/thermostat/properties"
>     temperature_state_template: "{{ value_json['targetTemperature'] | round(1) }}"
>     current_temperature_topic: "corridoiotermostato/thermostat/properties"
>     current_temperature_template: "{{states('sensor.thermostat_current_temperature')}}"
>     mode_command_topic: "corridoiotermostato/thermostat/set"
>     mode_state_topic: corridoiotermostato/thermostat/properties"
>     mode_state_template: >
>      {% set modes = { 'off':'heat', 'holiday':'home',  'eco':'away', 'auto':'auto', 'hold':'eco'} %}
>      {{ modes[value] if value in modes.keys() else 'off' }}
>     modes:
>       - "heat"
>       - "home"
>       - "away"
>       - "auto"
>       - "eco"
>     min_temp: 12
>     max_temp: 23
>     temp_step: 1.0
>     precision: 1.0
>     retain: true
>     initial: 16.0
> `

influenist, 2022-04-20:
Hi @Ruggero1912 @Pigotka , I'm about to buy 10 of these. Is there any update on the flashing possibilities without soldering directly the pins? I'm think of: 
TH213W for rooms with just underfloor heating
TH115W for rooms with underfloor heating and an AC
Or alternatively the newer version:
TH216B

BTW, the thermostats are developed by HeatCold brand which is a brand under the Xiaomi umbrella. Available under Tuya and Xiaomi Mija firmware. Quite strange that no one else is pushing on these. Please see their complete range here:
https://www.heatold.com/
and Chinese site with more products:
https://www.heatcold.cn/