klausahrenberg/WThermostatBeca #197: cannot get to flash mode

rudythebear, 2021-02-10:
hi,

Trying to flash a X7H thermostat, and failing.

https://www.ebay.co.uk/itm/Tuya-WiFi-LCD-Intelligent-Thermostat-Programmable-Temperature-Controller/392989344437?hash=item5b7ffd82b5:g:88sAAOSwLYVfkqKF

I have flashed other devices with tasmota in the past - such as sonoff basic and sonoff zigbee bridge, with no issues.

Used your excellent instructions and soldered the wires to the tyw3s module. 

however as soon as I power up the thermostat wakes up and goes "beep" and the background LED's light - which suggests to me that its NOT in flash mode (but at least Vcc and GND are correct! :)

tried: 
bringing GPIO0 to GND all the time.
connecting everything apart from Vcc, then connecting as the last item.
removing the Vcc from Enable
removing the GND from GPIO15

Also tried to bring Reset to GND (after the beep), but nothing - not even another beep.

any suggestions would be gratefully received. 

Rudy

rudythebear, 2021-02-10:
update: I cut the vcc connection on the PCB, connected the USB TTL dongle to the tyw3s and managed to flash the device!

waiting on a new solder tip to "re-connect" the cut pcb track.

rudy

kopakte, 2021-02-11:
I have a same problem. Can you post the image yours flashing setup?

rudythebear, 2021-02-11:
hi,

[image]

Please excuse the terrible soldering I had no small tips for the iron - I've removed the Vcc wire from the bottom left pin, so you can see the cut made in to the PCB - then attached Vcc (3.3v) from the TTL direct onto the daughter board. The wiring was the standard, tx-rx, rx-tx, I also grounded GPIO0 for the entire time.  

I used a hobby knife to cut the track, and to also scratch back some of the green insulation to allow me to solder a new bridge once I'd finished flashing.

rudythebear, 2021-02-11:
more notes: sorry is this is the wrong place for this..

connected everything back today, and I've upgraded to 1.20, selecting MK70GBH gets me most features -
* current temp
* target temp
* deviceOn
* schedulesMode

but there are a couple of issues - the thermostat can "handle" two weekend dates, the and the config only shows one, and I can't read the status of the switch - it's always heating.

klausahrenberg, 2021-02-20:
Good, you managed to flash the device. sometimes Tuya devices seems to resist :-)
Please look at MQTT for unknown MCU messages and report. Your device is not supported yet from my firmware. If the MK70 fits so far, it's good. However, if you send me the unknown messages then we will find the schedules message maybe and can implement the support

rudythebear, 2021-02-20:
hi, 
the fight was worth the effort! :)

okay - here's the data from my MQTT server.. hope it helps in some way. I also have the "instructions" that come with the device - do you want these to add to the library?

>thermostat_14466/clock/properties{"ntpServer":"pool.ntp.org","timeZoneServer":"http://worldtimeapi.org/api/ip","epochTime":1613845232,"epo                chTimeFormatted":"2021-02-20 18:20:32","validTime":true,"timezone":"Europe/London"}
>thermostat_14466/thermostat/properties{"idx":"thermostat_14466","ip":"192.168.1.127","alive":false}
>thermostat_14466/thermostat/properties/temperature14.90
>thermostat_14466/thermostat/properties/targetTemperature10.00
>thermostat_14466/thermostat/properties/deviceOntrue
>thermostat_14466/thermostat/properties/schedulesMode"auto"
>thermostat_14466/thermostat/properties/lockedfalse
>thermostat_14466/thermostat/properties/state"heating"
>thermostat_14466/thermostat/properties/statusMode"off"
>thermostat_14466{"notice":"MCU: 55 aa 03 07 00 05 01 01 00 01 01"}
>thermostat_14466{"notice":"MCU: 55 aa 03 07 00 08 03 02 00 04 00 00 00 95"}
>thermostat_14466{"notice":"MCU: 55 aa 03 07 00 05 04 04 00 01 01"}
>thermostat_14466{"notice":"MCU: 55 aa 03 07 00 05 05 04 00 01 00"}
>thermostat_14466{"notice":"MCU: 55 aa 03 07 00 05 08 01 00 01 00"}
>thermostat_14466{"notice":"MCU: 55 aa 03 07 00 05 0d 01 00 01 01"}
>thermostat_14466{"error":"Unknown MCU command: 55 aa 03 07 00 05 0d 01 00 01 01"}
>thermostat_14466{"notice":"MCU: 55 aa 03 07 00 05 10 05 00 01 00"}
>thermostat_14466{"error":"Unknown MCU command: 55 aa 03 07 00 05 10 05 00 01 00"}
>thermostat_14466{"notice":"MCU: 55 aa 03 07 00 08 14 02 00 04 ff ff ff f6"}
>thermostat_14466{"error":"Unknown MCU command: 55 aa 03 07 00 08 14 02 00 04 ff ff ff f6"}
>thermostat_14466{"notice":"MCU: 55 aa 03 07 00 08 15 02 00 04 00 00 00 3c"}
>thermostat_14466{"error":"Unknown MCU command: 55 aa 03 07 00 08 15 02 00 04 00 00 00 3c"}
>thermostat_14466{"notice":"MCU: 55 aa 03 07 00 05 19 04 00 01 00"}
>thermostat_14466{"error":"Unknown MCU command: 55 aa 03 07 00 05 19 04 00 01 00"}
>thermostat_14466{"notice":"MCU: 55 aa 03 07 00 05 1a 01 00 01 00"}
>thermostat_14466{"error":"Unknown MCU command: 55 aa 03 07 00 05 1a 01 00 01 00"}
>thermostat_14466{"notice":"MCU: 55 aa 03 07 00 05 1f 01 00 01 00"}
>thermostat_14466{"error":"Unknown MCU command: 55 aa 03 07 00 05 1f 01 00 01 00"}
>thermostat_14466{"notice":"MCU: 55 aa 03 07 00 05 29 04 00 01 03"}
>thermostat_14466{"error":"Unknown MCU command: 55 aa 03 07 00 05 29 04 00 01 03"}
>thermostat_14466{"notice":"MCU: 55 aa 03 07 00 05 2a 04 00 01 00"}
>thermostat_14466{"error":"Unknown MCU command: 55 aa 03 07 00 05 2a 04 00 01 00"}
>thermostat_14466{"notice":"MCU: 55 aa 03 07 00 24 2b 00 00 20 07 00 00 be 08 00 00 be 10 1e 00 78 15 00 00 be 16 1e 00 64 17 00 00 64 07 0a 00 be 0a 00 00 64"}
>thermostat_14466{"notice":"MCU: 55 aa 03 07 00 05 65 01 00 01 00"}
>thermostat_14466{"error":"Unknown MCU command: 55 aa 03 07 00 05 65 01 00 01 00"}
>thermostat_14466{"notice":"MCU: 55 aa 03 07 00 08 69 02 00 04 00 00 00 0a"}
>thermostat_14466{"error":"Unknown MCU command: 55 aa 03 07 00 08 69 02 00 04 00 00 00 0a"}
>thermostat_14466{"notice":"MCU: 55 aa 03 07 00 08 6b 02 00 04 00 00 00 23"}
>thermostat_14466{"error":"Unknown MCU command: 55 aa 03 07 00 08 6b 02 00 04 00 00 00 23"}
>thermostat_14466{"notice":"MCU: 55 aa 03 07 00 08 02 02 00 04 00 00 00 64"}

jpakusch, 2021-08-07:
Hi Everybody,
Can i Jump oboard this topic as well?

I have trouble flashing a too using an BHT-3000GALW

What i tried:
exactly following your soldering guide using an esp Programmer, you already guessed it, i was not able to get to flash mode.
So, i used the exact same setup with an ESP i had lying around (same pin layout). worked flawlessly.

Now, i desoldered the onboard ESP and soldered mine to the Board. Assembled the Device, put it to 230V, nothing Burst into Flames. So far, so good. But sadly, no AP containing "Thermostat" or anything similar appears.
Pushing the up arrow key, as this was the procedure in default configuration did not help either.

Afterwards, i tried to flash the ESP the device came with outside of the device, just the naked ESP(inspired by this Thread), this did not work either.
Any Ideas?