klausahrenberg/WThermostatBeca #198: Add reset function for no WiFi is possible?

alessandrosalafia, 2021-02-14:
It would be convenient to have a programmable function as it exposes that in case of wifi lack it reboots the device after a programmable time.

klausahrenberg, 2021-02-20:
Ok, your device didn't find a wifi and then it goes in AP mode for configuration after some minutes. You are suggesting that in this state, the device should reboot after some time to try the stored wifi credentials again?

alessandrosalafia, 2021-02-20:
> Ok, your device didn't find a wifi and then it goes in AP mode for configuration after some minutes. You are suggesting that in this state, the device should reboot after some time to try the stored wifi credentials again?

The functionality that it would be convenient to insert comes from a wifi coverage problem.  When the signal is weak the antennas of esp8266 lose the signal so I am forced to restart either the router or the thermostats.  If it were possible to force restart or even a network search to reconnect the device it would be very convenient.  Avoid continuous brutal reboots of devices.

DSF4ever, 2022-01-03:
Hi Gents.
I am struggeling with the same or at least a similar issue.
I am using a BECA BHT-8000 Heating Thermostat (I know it's not officially supported, but it works well with MQTT when configured with "Thermostat model: BHT-002, BHT-6000, BHT-3000 (floor heating)" )

I am loosing quite often (around 2-4 times a day) WIFI connection and the only way to get it work again is to power the BHT-8000 off and on.* (pls. see my workaround at the end)
To reboot the WIFI Access Point does not help. It's the BHT-8000 who falls into a coma.

Like Alessandro suggested, an automatic reset if WIFI gets lost would be cool.
To avoid an endless reset loop, may be in such a way:
WIFI lost -> reset (or a "WIFI reconnect" )
If WIFI OK -> DONE, else wait 40 seconds -> reset
If WIFI OK -> DONE, else wait 80 seconds -> reset
If WIFI OK -> DONE, else fallback in AP mode (which would be around 2 minutes after 3 resets)

And if this could be implemented as configurable option "yes, need that  / no, don't need that" it would be perfect (at least in my opinion :-) )

Thx for your thoughts!
DSF

* I only see this issue when the BHT-8000 is in our flat in Vienna, where the 2,4GHz WIFI is veeeery busy. Same BHT-8000 in "my lab" in our house in the middle of nowhere was connected for 14 days whithout a single connection loss.
I invested a lot of time in better WIFI (3 different WIFI APs, fixed channel vs automatic channel, channel width, etc... whatever my APs provided...), but could not improve the situation. Then I found this " https://github.com/letscontrolit/ESPEasy/issues/1987 " and I recognized if my issue is similar, I need to solve it in another way:
The power of my BHT-8000 is now controlled by a SONOFF WIFI switch. I'm using openHAB and ping the BHT-8000 every 30 seconds, if 2 pings get lost the SONOFF is switched off and switch on 3 seconds later. This "brutal reboots" brings the BHT-8000 back to live. Not very sophisticated, but this works now for a couple of month.