klausahrenberg/WThermostatBeca #203: Support for BHP-8000-US-MG?

hopkin, 2021-02-23:
Hi, can I request support for the BHP-8000? I've got one running 1.20m but as soon as I connect it to MQTT it spams out these errors:
"{"error":"Unknown MCU command: 55 aa 03 07 00 05 04 04 00 01 02"}"
"{"error":"Unknown MCU command: 55 aa 03 07 00 05 07 01 00 01 00"}"
"{"error":"Unknown MCU command: 55 aa 03 07 00 74 11 00 00 70 01 68 02 d0 01 e0 02 d0 02 b2 02 d0 03 2a 02 d0 01 68 02 d0 01 e0 02 d0 02 b2 02 d0 03 2a 02 d0 01 68 02 d0 01 e0 02 d0 03 fc 02 d0 05 28 02 6c 01 68 02 d0 01 e0 02 d0 02 b2 02 d0 03 2a 02 d0 01 68 02 d0 01 e0 02 d0 02 b2 02 d0 03 2a 02 d0 01 68 02 d0 01 e0 02 d0 03 fc 02 d0 05 28 02 6c 01 68 02 d0 01 e0 02 d0 02 b2 02 d0 03 2a 02 d0"}"

Please let me know if there's anything I can do on my end to help!

hopkin, 2021-02-23:
[image]

badger707, 2021-03-29:
Have not tried this firmware yet, but we had great success with BHT-8000 thermostat with this fork [here](https://github.com/fashberg/WThermostatBeca/issues/89)