klausahrenberg/WThermostatBeca #206: Support for "Hidden SSID" and "Static IP" address?

strizl, 2021-02-28:
Dear Klaus!

Thanks for your great firmware! I have it already setup on a couple of WiFi thermostats and I am very happy with.

Only 2 short questions regarding your FW:
Is there a possibilitiy to add support for a hidden SSID? My whole house automation communicates on an hidden SSID and it would be nice to add also my thermostats to it.
Next question would be to setup an fixed/static IP address in the config. Is there a possibility to or are you looking forward to add such a functionality in near future?

Would be great to get feedback from you!

Thanks a lot!
Kind regards,
strizl

wongcz, 2021-03-01:
+1 for static IP

Polomans, 2021-03-02:
So what's the problem?
`Backlog IPAddress1 192.168.50.41; IPAddress2 255.255.255.0; IPAddress3 192.168.50.41; IPAddress4 192.168.50.1; restart 1`
This does not work? I just haven't tried it myself.

wongcz, 2021-03-02:
There is no console accessible to write this Tasmota-like configuration

strizl, 2021-03-05:
Sorry, my fault:
I encountered hidden SSID works already fine!! Thx and sorry again!

But still +1 for STATIC IP ;-)