klausahrenberg/WThermostatBeca #208: Is this firmware compatible with all BHT-002 models?

ovizii, 2021-03-03:
I am specifically looking to buy a Moes BHT-002-GCLZB and would like to use your firmware. 

The search didn't yield any results for GCLZB hence my question.

GGW42, 2021-03-28:
Hi, I am not the owner or even an expert but this firmware is for the WIFI versions of this Thermostat, the "ZB" on the end of yours seems to mean "Zigbee".

This website seems to offer support using this thermostat via a Zigbee bridge to MQTT for Home Assistant et el.
https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/devices/BHT-002-GCLZB.md
I have no other knowledge of your unit or this github site, hope this helps

ovizii, 2021-03-29:
thanks for the hint, I guess I had a very blonde moment :-)

closed 2021-03-29 by ovizii