klausahrenberg/WThermostatBeca #211: ME108H - NC and NO swapped?

Rainerlan, 2021-03-26:
My ME108H works inverted:
I have NC Actors (Closing when not connected to line).
When my ME108H is not supposed to heat (set temperature below current temperature), the NC terminal gets switched to line (230V). 
Hence, the NC-actor opens and hence, the heating is ON (instead of OFF).
Note: The display symbol of ME108H for heating is "disabled" - as it should.
So, the logic of the terminals is inversed (when I connect NC actors to the NC terminal of ME108H).
After switching the terminals (NC actor connected to NO terminal), the heating works as expected.
Is it a bug of the TuyaMCU-code or could that be a bug of the esp8266 firmware? (I am using the latest beta of 1.20)
Am I thinking wrong here?
Anyways...Now I just connected my NC-actors to the NO terminal of the actors.
Once I get new ME108H's, I will check the terminal signal before flashing (I am too lazy to re-flash the stock FW for now)

o0v7r0o, 2021-03-28:
@Rainerlan 
first, the esp firmware has nothing to do with how the relay outputs are controlled, a separate chip does that
second, the way it operates in your description is exactly as it should -> NC - normally closed relay is connected to the live wire when the heating should be OFF.

For NC valves you should use NO outputs:
idle state -> NO relay is open -> valve is closed.
heating state -> NO relay is closed (shorted) -> valve is open.

Rainerlan, 2021-03-29:
Thanks @o0v7r0o for the confirmation. So I was wrong (altough aliexpress pictures do indicate that NO/NC on the thermostat imply the type of valve that is being connected to the terminals - and not the relais-behaviour of the thermostat).
But anyway - Thanks a lot for the confirmation and for taking the time to comment my question!
Seems after the time change this weekend I need to raise another question: Can the time be set via MQTT?

closed 2021-03-29 by Rainerlan