klausahrenberg/WThermostatBeca #215: Unknown MCU command 1.20m Beta Beca 6000

lokidaibel, 2021-05-12:
What is this ? Will it Brik my becas ? How to fix this ?

beca_maja {"error":"Unknown MCU command: 55 aa 01 07 00 05 68 01 00 01 01"}
beca_maja {"error":"Unknown MCU command: 55 aa 01 07 00 05 67 04 00 01 01"}
beca_hannah {"error":"Unknown MCU command: 55 aa 01 07 00 05 68 01 00 01 01"}
beca_hannah {"error":"Unknown MCU command: 55 aa 01 07 00 05 67 04 00 01 01"}
beca_bad {"error":"Unknown MCU command: 55 aa 01 07 00 05 68 01 00 01 01"}
beca_thermo_buer {"error":"Unknown MCU command: 55 aa 01 07 00 05 68 01 00 01 01"}
beca_thermo_buer {"error":"Unknown MCU command: 55 aa 01 07 00 05 67 04 00 01 01"}
beca_bad {"error":"Unknown MCU command: 55 aa 01 07 00 05 67 04 00 01 01"}
beca_maja {"error":"Unknown MCU command: 55 aa 01 07 00 05 68 01 00 01 01"}
beca_maja {"error":"Unknown MCU command: 55 aa 01 07 00 05 67 04 00 01 01"}
beca_hannah {"error":"Unknown MCU command: 55 aa 01 07 00 05 68 01 00 01 01"}
beca_hannah {"error":"Unknown MCU command: 55 aa 01 07 00 05 67 04 00 01 01"}
beca_thermo_buer {"error":"Unknown MCU command: 55 aa 01 07 00 05 68 01 00 01 01"}
beca_bad {"error":"Unknown MCU command: 55 aa 01 07 00 05 68 01 00 01 01"}
beca_thermo_buer {"error":"Unknown MCU command: 55 aa 01 07 00 05 67 04 00 01 01"}
beca_bad {"error":"Unknown MCU command: 55 aa 01 07 00 05 67 04 00 01 01"}
beca_maja {"error":"Unknown MCU command: 55 aa 01 07 00 05 68 01 00 01 01"}
beca_maja {"error":"Unknown MCU command: 55 aa 01 07 00 05 67 04 00 01 01"}
beca_hannah {"error":"Unknown MCU command: 55 aa 01 07 00 05 68 01 00 01 01"}
beca_hannah {"error":"Unknown MCU command: 55 aa 01 07 00 05 67 04 00 01 01"}
beca_thermo_buer {"error":"Unknown MCU command: 55 aa 01 07 00 05 68 01 00 01 01"}
beca_thermo_buer {"error":"Unknown MCU command: 55 aa 01 07 00 05 67 04 00 01 01"}
beca_bad {"error":"Unknown MCU command: 55 aa 01 07 00 05 68 01 00 01 01"}
beca_bad {"error":"Unknown MCU command: 55 aa 01 07 00 05 67 04 00 01 01"}
beca_maja {"error":"Unknown MCU command: 55 aa 01 07 00 05 68 01 00 01 01"}
beca_maja {"error":"Unknown MCU command: 55 aa 01 07 00 05 67 04 00 01 01"}
beca_hannah {"error":"Unknown MCU command: 55 aa 01 07 00 05 68 01 00 01 01"}
beca_hannah {"error":"Unknown MCU command: 55 aa 01 07 00 05 67 04 00 01 01"}
beca_thermo_buer {"error":"Unknown MCU command: 55 aa 01 07 00 05 68 01 00 01 01"}
beca_thermo_buer {"error":"Unknown MCU command: 55 aa 01 07 00 05 67 04 00 01 01"}
beca_bad {"error":"Unknown MCU command: 55 aa 01 07 00 05 68 01 00 01 01"}
beca_bad {"error":"Unknown MCU command: 55 aa 01 07 00 05 67 04 00 01 01"}
beca_maja {"error":"Unknown MCU command: 55 aa 01 07 00 05 68 01 00 01 01"}
beca_maja {"error":"Unknown MCU command: 55 aa 01 07 00 05 67 04 00 01 01"}
beca_hannah {"error":"Unknown MCU command: 55 aa 01 07 00 05 68 01 00 01 01"}
beca_hannah {"error":"Unknown MCU command: 55 aa 01 07 00 05 67 04 00 01 01"}
beca_thermo_buer {"error":"Unknown MCU command: 55 aa 01 07 00 05 68 01 00 01 01"}
beca_thermo_buer {"error":"Unknown MCU command: 55 aa 01 07 00 05 67 04 00 01 01"}
beca_bad {"error":"Unknown MCU command: 55 aa 01 07 00 05 68 01 00 01 01"}
beca_bad {"error":"Unknown MCU command: 55 aa 01 07 00 05 67 04 00 01 01"}
beca_maja {"error":"Unknown MCU command: 55 aa 01 07 00 05 68 01 00 01 01"}
beca_maja {"error":"Unknown MCU command: 55 aa 01 07 00 05 67 04 00 01 01"}

Gopten, 2021-05-20:
Go to the web page of your Beca     http://ipaddressofbeca/thermostat?  and untick send mcu commands

peopeo1, 2021-05-21:
I have same problem.
{"error":"Unknown MCU command: 55 aa 01 07 00 05 68 01 00 01 01"}   every 2 second on mqtt 
send mcu commands is off
1.20m beta
BHT-002

peopeo1, 2021-06-09:
In latest version 1.20n its ok too, on BHT-002