klausahrenberg/WThermostatBeca #223: Full working YAML for HASS

dusan-ivanco, 2021-07-20:
```
climate:
  - platform: mqtt
    name: Thermostat
    temperature_command_topic: climate/thermostat/command
    temperature_command_template: >
      {{ {'targetTemperature':value} | to_json }}
    temperature_state_topic: climate/thermostat/properties
    temperature_state_template: >
      {% if value_json is defined and value_json['targetTemperature'] is defined %}
        {{ value_json['targetTemperature'] }}
      {% else %}
        20
      {% endif %}
    current_temperature_topic: climate/thermostat/properties
    current_temperature_template: >
      {% if value_json is defined and value_json['temperature'] is defined %}
        {{ value_json['temperature'] }}
      {% else %}
        20
      {% endif %}
    mode_state_topic: climate/thermostat/properties
    mode_state_template: >
      {% if value_json is defined and value_json['deviceOn'] is defined %}
        {% set deviceOn = value_json['deviceOn'] %}
      {% else %}
        {% set deviceOn = false %}
      {% endif %}

      {% if value_json is defined and value_json['schedulesMode'] is defined %}
        {% set schedulesMode = value_json['schedulesMode'] %}
      {% else %}
        {% set schedulesMode = 'off' %}
      {% endif %}

      {% if deviceOn == true %}
        {% if schedulesMode == 'auto' %}
          auto
        {% else %}
          heat
        {% endif %}
      {% else %}
        off
      {% endif %}
    mode_command_topic: climate/thermostat/command
    mode_command_template: >
      {% if value == 'auto' %}
        {{ {'deviceOn':true,'schedulesMode':'auto'} | to_json }}
      {% endif %}

      {% if value == 'heat' %}
        {{ {'deviceOn':true,'schedulesMode':'off'} | to_json }}
      {% endif %}

      {% if value == 'off' %}
        {{ {'deviceOn':false,'schedulesMode':'off'} | to_json }}
      {% endif %}
    modes:
      - "auto"
      - "heat"
      - "off"
    temperature_unit: C
    temp_step: 0.5
    min_temp: 5
    max_temp: 35
    initial: 20
    retain: true
    qos: 0
```

WildeRNS, 2021-09-23:
Great!
Maybe someone can share schedule config?

ikke-t, 2022-09-09:
Thanks, I got this working with my setup too. However HA is saying this is deprecated, and will get disabled soon. Do you know how it should be modified?

[image]

ikke-t, 2022-09-09:
OK, fixed this. This is the current setup of mine. I added also the device section and unique id, so it becomes possible to edit it in GUI:

```
mqtt:
  climate:
    - name: Pukuhuoneen lattialämmitys
      unique_id: termari-pukuhuone
      device:
        configuration_url: http://192.168.1.5/config
        manufacturer: Tuya
        model: ME81AH
        suggested_area: Pukuhuone
        connections: [["mac", "DE:AD:BE:EF:FF:FF"]]
      temperature_command_topic: termari-ph/thermostat/set
      temperature_command_template: >
        {{ {'targetTemperature':value} | to_json }}
      temperature_state_topic: termari-ph/thermostat/properties
      temperature_state_template: >
        {% if value_json is defined and value_json['targetTemperature'] is defined %}
          {{ value_json['targetTemperature'] }}
        {% else %}
          20
        {% endif %}
      current_temperature_topic: termari-ph/thermostat/properties
      current_temperature_template: >
        {% if value_json is defined and value_json['temperature'] is defined %}
          {{ value_json['temperature'] }}
        {% else %}
          20
        {% endif %}
      mode_state_topic: termari-ph/thermostat/properties
      mode_state_template: >
        {% if value_json is defined and value_json['deviceOn'] is defined %}
          {% set deviceOn = value_json['deviceOn'] %}
        {% else %}
          {% set deviceOn = false %}
        {% endif %}
  
        {% if value_json is defined and value_json['schedulesMode'] is defined %}
          {% set schedulesMode = value_json['schedulesMode'] %}
        {% else %}
          {% set schedulesMode = 'off' %}
        {% endif %}
  
        {% if deviceOn == true %}
          {% if schedulesMode == 'auto' %}
            auto
          {% else %}
            heat
          {% endif %}
        {% else %}
          off
        {% endif %}
      mode_command_topic: termari-ph/thermostat/set
      mode_command_template: >
        {% if value == 'auto' %}
          {{ {'deviceOn':true,'schedulesMode':'auto'} | to_json }}
        {% endif %}
  
        {% if value == 'heat' %}
          {{ {'deviceOn':true,'schedulesMode':'off'} | to_json }}
        {% endif %}
  
        {% if value == 'off' %}
          {{ {'deviceOn':false,'schedulesMode':'off'} | to_json }}
        {% endif %}
      modes:
        - "auto"
        - "heat"
        - "off"
      temperature_unit: C
      temp_step: 0.5
      min_temp: 5
      max_temp: 35
      initial: 20
      retain: true
      qos: 0
```