klausahrenberg/WThermostatBeca #226: DLX-LH1 schedulesMode states "off" and "auto" are switched - no setpoint via mqtt possible

mVik85, 2021-08-24:
Hi @klausahrenberg!

Just tried to integrate my DLX-LH1 to Domoticz  (via mqtt through Node-RED), and found that the "off" and "auto" schedulesMode states are switched in 1.20n.
So the device returns "off" when it is set to scheduled mode on the touch panel and vica versa. I assume there is a condition check for this state, because as a side effect it is not possible to change the "targetTemperature" when in "off" (manual) mode. The device first returns the new set value, but then immediately the previous one. 
Possible workaround is to set the device to "auto", send the "targetTemperature" then switch the schedulesMode back to "off" and with this the new setpoint is set.

Could this be fixed alone by modifying the WThermostat_DLX_LH01.h?
Unfortunately I don't know C++, so i have no further clue:)

Thanks!

NoFace3D, 2021-09-05:
I can confirm this issue. Modes off and auto are switched.

jdomenechg, 2022-01-02:
Same problem here, modes off and auto are switched in DLX-LH1.

More over the schedules are not saved when changing them from the web interface. :-(