klausahrenberg/WThermostatBeca #228: BHT-002-GCLW DLX-LH01 board Unknown MCU commands

NoFace3D, 2021-09-05:
Hello,

I am reporting Uknown MCU commands for BHT-002-GCLW with DLX-LH01 board:

{"error":"Unknown MCU command: 55 aa 03 07 00 74 11 00 00 70 01 68 00 c8 02 b2 00 96 03 2a 00 96 04 1a 00 c8 01 68 00 c8 02 b2 00 96 03 2a 00 96 04 1a 00 c8 01 68 00 c8 02 b2 00 96 03 2a 00 96 04 1a 00 c8 01 68 00 c8 02 b2 00 96 03 2a 00 96 04 1a 00 c8 01 68 00 c8 02 b2 00 96 03 2a 00 96 04 1a 00 c8 01 68 00 c8 02 b2 00 96 03 2a 00 96 04 1a 00 c8 01 68 00 c8 02 b2 00 96 03 2a"}
{"error":"Unknown MCU command: 55 aa 03 07 00"}
{"error":"Unknown MCU command: 55 aa 03 07 00 08 0c 02"}

They are logged quite frequently almost every second.

NoFace3D, 2021-09-05:
release 1.20n

carvan, 2021-10-17:
Same here. Just upgraded to 1.19 and I'm getting a lot of these messages and I'm unable to control it via Home assistant

titiviking, 2022-03-04:
I have the same, and on top it seems that the topics used for discovery are not the same as MQTT HA used for auto-discovery. Hence no way to make it work...