klausahrenberg/WThermostatBeca #231: ME81

Dragon0501, 2021-09-10:
After flashing newest Beta 1.20n the device isn't reachable. Sometimes it says Magic Bit is wrong and sometimes flashing works but every time I have to reflash an older Version by FTDI. 
The target temperature cannot be written just the changed temeprature from the device is delivered by MQTT,

JMyrng, 2021-10-02:
Version 1.20n works for me with the ME81H.16. 

I flashed 1.19 via FTDI and then changed to the latest version via OTA. As described, I had to set up the whole device again. 

With the ME81H, I can change the temperature manually in automatic mode, then both the manual and automatic symbols are displayed. However, the new temperature is not transmitted via MQTT. But everything else is working fine.

roleoroleo, 2021-11-02:
Could you explain me how did you flash Tuya module?
Have you unsoldered it?

Dragon0501, 2021-11-02:
No I didn't unsoldered it. Just soldered the cables and flashed it...

JMyrng, 2021-11-02:
> Could you explain me how did you flash Tuya module? Have you unsoldered it?

No, I've done it with this explanation: https://github.com/klausahrenberg/WThermostatBeca/blob/master/Flashing.md#2-connection-to-device-for-flashing

roleoroleo, 2021-11-02:
Thank you.
I will try it.

roleoroleo, 2021-11-02:
Success, thank you.
Hw mod for relay state done.
Everything is working fine.

roleoroleo, 2021-11-06:
When I try to send commands from an mqtt client, sometimes me81h receives and understands the command, sometimes ignores it.
I'm sure the mqtt server got it.
Is it the same as your problem?

JMyrng, 2021-11-06:
> Is it the same as your problem?

No, for me, when manually overwriting the automatic temperature, the change is not sent via MQTT. That is, from the thermostat to the MQTT server.