klausahrenberg/WThermostatBeca #241: BHT-002 with TUYA WB3S Module

PhilipTS, 2021-11-23:
Hello All,

Just to inform you all that I just received a MOES BHT-002 and it has a TUYA WB3S module for WiFi instead of an ESP module

https://www.moeshouse.com/collections/water-heating-thermostat/products/wifi-white-thermostat-temperature-controller-lcd-touch-screen-backlight-for-water-floor-heating-works-with-alexa-google-home-3a

https://developer.tuya.com/en/docs/iot/wb3s-module-datasheet?id=K9dx20n6hz5n4

Will try to upload pictures later ... and I will try to figure out if we can swap the module?

PhilipTS, 2021-11-23:
And I just realized that I always assumed that they have an ESP module until I received mine.

Has anybody flashed WThermostatBeca to these TUYA KWH-WB3S (v1.1) modules?

PhilipTS, 2021-11-23:
Attached images ...

[image]
[image]

shakin89, 2021-11-24:
I've bought one of these thermostat (becasmart BAC-002-ELW) thinking it would have the tywe chip, but mine has the wb3s chips mounted.
Is there a way to make it work with home assistant? I've seen someone desolder the chip and solder an eps01 that has the same pinout, but i have to replace ten of these thermostat and wouldn't like to make this operation for all.

sematuszewski, 2021-11-25:
I was lucky. I ordered 8 BHT-002 GBLW, today I picked up 6 and all on TYWE3S. The pcb board marked as ZWX-GB_2020.10.17.
You can always try to replace the controller on ESP, at the moment, unfortunately, it's probably the only option.

PhilipTS, 2021-11-25:
I already flashed WThermostatBeca on an ESP module that I had and hopefully I will find some time this weekend to remove the WB3S and pull some wires to my ESP module and see if it works.

I already ordered 2 TYWE3S from aliexpress as i am afraid that soon all Tuya Thermostats will come with the WB3S.
I just home that they didn't change anything on the MCU side of things. I guess I will find out when I wire up my ESP module.

shakin89, 2021-11-25:
Thanks for your answers. 
Philip let us know what happens. 
I have two spare esp12-e that i will solder replacing the wb3s and try to flash this firmware or the esphome. I'm trying to figure if esphome is able to manage the tuya thermostat.

PhilipTS, 2021-11-26:
I had some of these in my toolbox, so I flashed WThermostatBeca on one of them.

[image]

I removed the WB3S and wired up my ESP module ... (just the VCC, GND, TX, RX) ... did damage some tracks on the PCB :(
Everything looks to be working fine, but it's the first time that I use WThermostatBeca.
I played around with the thermostat and I checked the different properties.
I saw these values change: alive, temperature, targetTemperature, deviceOn, schedulesMode, locked.
The state property was always set to heating.

Is there a way to see if the actual heating is ON or OFF, i mean the actual relay status.

thermostat_493653/thermostat/properties:
{
  "idx": "thermostat_493653",
  "ip": "192.168.73.221",
  "alive": true,
  "firmware": "1.20n",
  "temperature": 19,
  "targetTemperature": 14.5,
  "floorTemperature": 0,
  "deviceOn": true,
  "schedulesMode": "off",
  "locked": false,
  "state": "heating"
}

thermostat_493653/thermostat/properties/schedules"

{
  "w1h": "06:00",
  "w1t": 22,
  "w2h": "08:00",
  "w2t": 15,
  "w3h": "11:30",
  "w3t": 15,
  "w4h": "13:30",
  "w4t": 15,
  "w5h": "17:00",
  "w5t": 22,
  "w6h": "22:00",
  "w6t": 15,
  "a1h": "06:00",
  "a1t": 20,
  "a2h": "08:00",
  "a2t": 20,
  "a3h": "11:30",
  "a3t": 20,
  "a4h": "13:30",
  "a4t": 20,
  "a5h": "17:00",
  "a5t": 20,
  "a6h": "22:00",
  "a6t": 15,
  "u1h": "06:00",
  "u1t": 20,
  "u2h": "08:00",
  "u2t": 20,
  "u3h": "11:30",
  "u3t": 20,
  "u4h": "13:30",
  "u4t": 20,
  "u5h": "17:00",
  "u5t": 20,
  "u6h": "22:00",
  "u6t": 15
}

PhilipTS, 2021-11-26:
I see that on the screen I can see when the heating/relay is ON from the smoke that appears over the chimney (green circle below).

Is there a way to get this info from the MCU in a message over MQTT?

[image]

PhilipTS, 2021-11-26:
To answer my own question ... details in the post below ...

https://github.com/klausahrenberg/WThermostatBeca/issues/17#issuecomment-552078026

More soldering ... i don't know if my PCB will survive my sloppy/impatient hands :)

closed 2021-11-26 by PhilipTS

dwarning, 2021-12-17:
I just received an BECA BHT-002GALW and got the same wifi module WB3S. The time seems over for simple tasmota flashing. I dread it to remove the module because I haven't hot air de-solder tool.

ilkkaolavas, 2023-10-06:
[image]

Hi I have successfully used WThermostatBeca in 12 BHT-002 thermostats but now I received WB3S chipsed based version. I asked from official moeshouse shop and I got message that they have stopped producing TYWE3S based devices. I don't know if this is true but I think that supporting another chipset versions could be possible. ESPhome just started to use LibreTiny. Could it be possible to implement WB3S support also for WThermostatBeca ?

Bl00d-B0b, 2024-02-13:
Hello,

Yes, and it's works perfect. You can cut tuya with tuya-cloudcutter and install esphome
[https://devices.esphome.io/devices/Beca-Thermostat](url)

g6094199, 2024-02-13:
> Hello,
> 
> Yes, and it's works perfect. You can cut tuya with tuya-cloudcutter and install esphome [https://devices.esphome.io/devices/Beca-Thermostat](url)

...or alternatively flash [OpenBK](https://github.com/openshwprojects/OpenBK7231T_App) onto it.