klausahrenberg/WThermostatBeca #244: Unable to flash Moes GALWW-N thermostat

dieterlind, 2021-12-04:
Hello!

I am stuck since days and weeks in trying to flash on of my Moes GALWW-N thermostat which is basically a BAC-002ALWW (https://www.moeshouse.com/collections/smart-thermostat/products/wifi-central-air-conditioner-thermostat-temperature-controller-2-pipe-4-pipe-3-speed-fan-coil-unit-work-with-alexa-google-home).
The hardware board says : ZWX-GA/GC-CONTROL 2020.10.17
There is definitely the TYWE3S chip there.
I connected as mentioned here and also tried some different combinations of GPIO0, RST found in the internet with regards to flashing TYWE3S-chip.
All without success.
As soon as I connect the usb-ttl-converter the thermostate "boots" into normal mode.
How should it behave if in flash mode?? Can I somehow check if in flash mode??
Also tried to get information by trace of esptool.py .... always says no communication
Do I have to reset also the MCU?? If yes, which one is it on the device and which pin to use??
The right chip is labeled as TM1621B.

[image]

Hope someone can help me in flashing this device!

Thanks!

shakin89, 2021-12-04:
have you already checked this guide on how to flash?
https://www.youtube.com/watch?v=imKZbhJ8lvU

dieterlind, 2021-12-04:
Hi!
Thanks for the link! Didn't know the video. But actually try it with a 3d-printed-adapter and pogo pins ... unfortunately without success until now ... 
the part with reset on the second chip is making me worrying as I don't know where it is on my board .... 
and with esptool.py I don't get any output ..

shakin89, 2021-12-04:
That's a tywe3s chip, if you look for the pinout on google, you will find where is reset on your board.

dieterlind, 2021-12-05:
Yes, true .. but as I understood in the other discussions (and also in your video) I have to reset the MCU ....

shakin89, 2021-12-05:
if you look for the tywe pin out, you will find that there is a pin called RST or Reset, you have to pull it to GND to have the flash mode enabled

dieterlind, 2021-12-09:
Hi!
Actually I am really stuck .... 
I guess I tried now all and every combination ... but somehow it doesn't work

What I did:
connect ESP VCC to TTL 3.3V
connect ESP EN to TTL 3.3V
connect ESP RST to TTL GND
connect ESP GND to TTL GND
connect ESP GPIO15 to TTL GND
connect ESP GPIO0 to TTL GND
connect ESP TX to TTL RX
connect ESP RX to TTL TX
connect UTF87012 RST (Pin 4) to TTL GND

Tried also not to connect EN, RST, GPIO 15 -> no success
When not connecting UTF87012 RST (Pin 4) to TTL GND the device obviously boot into normal mode ... when connected the sceen stay dark -> I guess this is a good sign to ensure that RX/TX are not used

I also tried is one of my sonoffs: connected VCC/GND/GPIO0/RX/TX wit the same cables -> work immediately

How can I get the boot log messages?? Maybe this would give me a hint.
[image]

[image]
[image]

tsrubar, 2021-12-09:
Hi, i have not been able to flash with same usb converter.
I succeeded using NodeMCU. You can find related issue  [https://github.com/fashberg/WThermostatBeca/issues/115#issuecomment-838277102](here)

dieterlind, 2021-12-09:
Hi! Thanks for the link ... I didn't know this one.
The hint with the baudrate was good ... now I get at least an output of the boot process (explanation of boot codes: https://riktronics.wordpress.com/2017/10/02/esp8266-error-messages-and-exceptions-explained/)

My output:

 ets Jan  8 2013,rst cause:2, **boot mode:(3,6)**

load 0x40100000, len 2592, room 16
tail 0
chksum 0xf3
load 0x3ffe8000, len 764, room 8
tail 4
chksum 0x92
load 0x3ffe82fc, len 676, room 4
tail 0
chksum 0x22
csum 0x22

2nd boot version : 1.7(5d6f877)
SPI Speed : 40MHz
SPI Mode : QIO
SPI Flash Size & Map: 8Mbit(512KB+512KB)
jump to run user1 @ 1000

So this means that bootmode (3,6) is not really good, or??

As mentioned on the ESP GPIO 15 and GPIO0 are connected to ground, GPIO2 is not connected.
It doesn't make any difference if I connect GPIO15 and/or GPIO0 ... the bootmode always stays 3,6 (sometimes 3,7).

Also the output with the right baudrate is readable ... do we still think the usb converter is the issue??