klausahrenberg/WThermostatBeca #246: MH102H.31 WIFI

AlbertWeterings, 2021-12-24:
I have a new version of the MH102H
Equipment Model: 102
So+C108ftware release: 1.0.0.0
It has a Wifi Module WBR3 instead of TYWE3S

I removed the origional WIFI module WBR3 and replaced it with a ESP12 it is more or less pin compatible. After that I flashed serveral versions of the WThermostatBeca. So far the ESP12 is working from the touchscreen I can enable and disable Wifi the radio is switching on and off.

This is the only thing working. It looks like there is no communication between the MCU and the ESP12. I tried to listen on the serial port TX(mcu) to RX(esp12) there is no baud rate that gives me anything readable.

Is there anyone who has this device working, knows the baud rate or knows a way to figure out how to capture the communication between mcu and ESP?

Fran6u, 2021-12-27:
> I have a new version of the ME102H Equipment Model: 102 So+C108ftware release: 1.0.0.0 It has a Wifi Module WBR3 instead of TYWE3S
> 
> I removed the origional WIFI module WBR3 and replaced it with a ESP12 it is more or less pin compatible. After that I flashed serveral versions of the WThermostatBeca. So far the ESP12 is working from the touchscreen I can enable and disable Wifi the radio is switching on and off.
> 
> This is the only thing working. It looks like there is no communication between the MCU and the ESP12. I tried to listen on the serial port TX(mcu) to RX(esp12) there is no baud rate that gives me anything readable.
> 
> Is there anyone who has this device working, knows the baud rate or knows a way to figure out how to capture the communication between mcu and ESP?
> 
> Update 1: I investigated a bit more and found the baud rate the MCU is using to communicate, the shortest pule is roughly 100µs -- this is 1/100*106 or 10,000bps -- this is obviously not a real baud-rate, but it is really close to 9,600bps. So I go for 9600 baud
> 
> It might be the way it communicates is different than in the program I see the WiFi symbol is blinking but WiFi is connected I can get to http://{IP}/things/thermostat/properties/temperature where the temperature stay's rock-solid at 0.00
> 
> I think the best way to proceed is to place back the original WiFi module and try to capture RAW communication. Anyone a Idea how-to?
> 
> Update 2: It works finally. The solution wat easy after replacing the WIFI module WBR3 with a ESP2 the only thing needed was adding a wire from GPIO 2 to the point on the PCB where the led (D10) is connected to the R34. This junction is also connected to GPIO14 of the ESP12
> 
> GPIO2 of the ESP12 becomes low when a WiFi network is connected. GPIO14 of the ESP12 needs to be low to have the MCU be aware WiFi is connected.

Hi. Can I ask if you're controlling the thermostat remotely via HA and if you could kindly advise me on how to achieve this. I got the ME102H wifi chip as well. Thanks

AlbertWeterings, 2021-12-27:
@Fran6u just to be clear the ME102H is the thermostat the wifi module inside has printed on it WBR3 This module is not a ESP8266 based module where @klausahrenberg has created this software for. So if yours has the same module (WBR3) inside you need to replace it with a TYWE3S or ESP12 module. For the ESP12 module you need to add the additional wire as I mentioned in my update: 2

I do not use HA. I don't need/want a web interface to control the thermostat. My goal is to use Node-Red to relay messages from the thermostat to a opentherm module controlling the central heating in my home. At the moment I use a simple ESP1 module with a DS18B20 temp sensor to do so. I ordered this ME102H thermostat after I found the work of @klausahrenberg the device looks nice and it just gives me a little bit more manual control with touch screen on the wall. Unfortunately the software in this thermostat is not giving me all data I would like to have from it at this point but that is something that can be worked on.

Fran6u, 2021-12-27:
@AlbertWeterings. Thanks for your prompt response. I have a few esp12 I can use in this project. And yes, the devise really does look nice; and for a fraction of the cost of other well known brands, it would be sweet to have full access to all it's capabilities. Thanks again for your response and good luck with your efforts.

PhilipTS, 2021-12-28:
Yesterday I received a ME102H.31 and replaced the WBR3 with a ESP8266 based module with WThermostatBeca 1.24beta.
The WiFi icon is flashing on the thermostat screen BUT the date/time are correct so I guess the ESP module set it?
I soldered all the pins of the ESP module on the PCB.

@Fran6u all you did was add "a wire from GPIO 2 to the point on the PCB where the led (D10) is connected to the R34"
with the standard bin files from here? what version of WThermostatBeca bib you use?

You REALLY saved me a lot of time (I would not be able to trouble shoot like you did).

PhilipTS, 2021-12-28:
[image]

@AlbertWeterings I see that next to D10 I have R43 ... you mention R34 and with my bad eyes I can't find R34 ... please help?

PhilipTS, 2021-12-28:
And to answer my own question ... i checked with my multi-meter and the point between D10 and R43 is connected to GPIO 14.
So I connected a wire between GPIO 14 and GPIO 2 and all is working fine now.

AlbertWeterings, 2021-12-28:
@PhilipTS The wire makes the WiFi icon stop flashing. (It took me two day's to figure out what was happening ;-) )There is one thing I could not check anymore. Did GPIO2 already have a connection to the MCU? Did you by acident take a picture of the board with the WiFi module removed? I tried to sniff on the communication between MCU and WiFi module that is very hard as the MCU is repeating almost ever 2 seconds the same set of commands. I think that can't be right. I wanted to start working on the issues I reported in [https://github.com/klausahrenberg/WThermostatBeca/issues/247](url) But that is undoable with this flood on data being send from the mcu

PhilipTS, 2021-12-28:
@AlbertWeterings 
I have already destroyed 2 BHT-002s with my weak/bad soldering skills that I didn't think of taking many pictures.
So no i don't have any with the module off. But with the wire between GPIO 2 & 14 everything is working fine.
I wish I had the skills, equipment & time to troubleshoot in such depth.

AlbertWeterings, 2021-12-28:
@PhilipTS Well in my case it is not always going right ;-) I destroyed the circuit creating 3v from the 12v supply to the WIFI module. Had to replace it with a different regulator. I think I have to order a new thermostat to experiment with the original module on it to monitor the communication between MCU and Module. But first we need @klausahrenberg his help as it seams I have no luck getting to have PlatformIO working so I can't build new test versions of the software.

PhilipTS, 2021-12-29:
@AlbertWeterings

I don't really know what I am doing but I am trying. So I flashed Tasmota 9.1.0.
Tasmota 10.1 would hang when I configured the TuyaMCU.

I enabled weblog 4 and I got the attached logs.
[tasmota_TuyaMCU_weblog4.txt](https://github.com/klausahrenberg/WThermostatBeca/files/7789101/tasmota_TuyaMCU_weblog4.txt)

Will try to configure everything in tasmota (I have some more reading to do).
Just thought I would upload the log file as it might help others in trouble shouting?

AlbertWeterings, 2021-12-29:
@PhilipTS it looks like this tasmoto version is only sending you all information from the MCU in raw format. This can be handy if you want to create your own program to process this date but from my perspective staying with the software @klausahrenberg has created makes your work more easy. In the mean time I forked this software and made some changes to, it is now working more as I like it. Still working on getting the relay status in the status message being broadcasted.

PhilipTS, 2021-12-29:
Have you seen this issue?

https://github.com/klausahrenberg/WThermostatBeca/issues/17

It mentions that the relay state is not sent by the MCU and we need a hardware mob to get it.
Have you figured out the pins for this mod?

Or are you going the other way like fashberg and calculating the state based on the temperatures.
https://github.com/fashberg/WThermostatBeca/

AlbertWeterings, 2021-12-29:
@PhilipTS No I tried to get the relay state from the MCU's output without modifying the hardware. In my thermostat it is working, when the flame on the display goes on I get the message Heater: true in the MQTT output. The only thing I can't test is if the relay is actually switching as I broke the 12V supply. Can you load [this binary](https://github.com/AlbertWeterings/WThermostatBeca/releases/download/V1.0.1-alpha/WThermostat_1.0.1_alpha.bin) and check please?

PhilipTS, 2021-12-29:
@AlbertWeterings 

Just loaded WThermostat_1.0.1_alpha.bin and I can see the Heater property change to true/false in sync with the small flame.
I can also here the relay click when we have a change.
Great Work !!!

[image]

PhilipTS, 2021-12-29:
@AlbertWeterings

I don't know what you did but it looks like you fixed the below issue ??? !!!

https://github.com/klausahrenberg/WThermostatBeca/issues/239

Will continue testing

PhilipTS, 2021-12-29:
Unfortunately issue 239 is not solved ...

AlbertWeterings, 2021-12-29:
I will have a look at #239 in the morning. Good to know so far what I made is working. Thanks for testing.

plush, 2023-03-24:
@AlbertWeterings Thank you for your investigation on this. I just bought an ME102H that came with a WBR3 module. I can confirm that after swapping this for an ESP-12F and wiring GPIO 2 to GPIO 14, things work perfectly.

The hardware mod to wire one pin to another works, but it does not feel very elegant. It would seem the correct solution is to modify the software, so that it pulls GPIO 14 low instead of / in addition to GPIO 2 (AFAICT, GPIO 2 isn't wired to anything in this hardware revision). WThermostatBeca uses the WAdapter libary and its WNetwork class actually has the ability to trigger a GPIO pin when WiFi connects. It is hard-coded to pulse the pin every 500ms to make an LED blink [1], but either changing that or simply toggling the pin directly via digitalWrite() should work.

[1] https://github.com/klausahrenberg/WAdapter/blob/806fab758acdf7123d787a9c00efc2bbde7ce217/src/WNetwork.h#L123