klausahrenberg/WThermostatBeca #247: ME102H Status message

AlbertWeterings, 2021-12-27:
The ME102H is sending every 5 minutes a status message like below.
`{"idx":"thermostat_1122403","ip":"192.168.1.182","alive":true,"firmware":"1.24beta","temperature":21.00,"targetTemperature":19.00,"floorTemperature":0.00,"deviceOn":true,"schedulesMode":"off","locked":false,"sensorSelection":"internal"}"`

This message is giving some useful information but not all information needed to use it as a good thermostat for an opentherm driven heating device.

We have:
the current temperature
target Temperature
the current floor Temperature

In the thermostat we can also put the Temperature Limit and floor Temperature Limit with these values we can set the heating device limits via opentherm so it would be usefull to have these values in the status message.

I also noticed that the target Temperature is not changing to the temperature set for Away and Hold mode. There is also nothing in the status message telling these modes are activated.

@klausahrenberg :
Can you add these values to the status message?
- Temperature Limit
- Floor Temperature Limit
- Leave/Away Temperature
- Hold/Maintain Temperature
- Hold/Maintain Time

AlbertWeterings, 2021-12-29:
I created a [fork](https://github.com/AlbertWeterings/WThermostatBeca) of this repository with some changes:

It was not possible to extract all above requested info from the MCU's output messages. What I could do was adding the temperature limits for the heater. These are now in the fields MinTemperatureHeater and MaxTemperatureHeater how this is done is explained in detail below. With these values it is possible to steer your heater in a way it is working as efficient as possible in terms of gas usage. Additional I changed the scheduler behavior as it was not possible to overwrite scheduled values manually without disabling the scheduler in the menu. 

What is changed:
Added output for the following items:
MinTemperatureHeater:
55 AA 03 07 00 08 1A 02 00 04 00 00 00 05 36 Min 05 Celcius (Command ID 1A)
55 AA 03 07 00 08 1A 02 00 04 00 00 00 14 45 Min 20 Celcius (Command ID 1A)
MaxTemperatureHeater:
55 AA 03 07 00 08 13 02 00 04 00 00 00 23 4D Max 35 Celcius (Command ID 13)
55 AA 03 07 00 08 13 02 00 04 00 00 00 5D 87 Max 93 Celcius (Command ID 13)
55 AA 03 07 00 08 13 02 00 04 00 00 00 5E 88 Max 94 Celcius (Command ID 13)

Changed the behavior of the scheduler:
Make sure manual overwrite is always working even if the scheduler is on AUTO
this way you can temporary manually overwrite the scheduled temperature.
The scheduled program will take over at the next scheduled time block.
Away during scheduler AUTO is now setting the temperature to the value set in "Leave"
--- The scheduler does not overwrite Away mode ---
Hold during scheduler AUTO is now setting the temperature to the value set in "Maintain"
--- The scheduler does not overwrite Hold mode it will continue after the Hold timer has counted down ---

Rainerlan, 2022-01-25:
@AlbertWeterings: I have tried several commands with my ME108H on your fork, however I can't get the following to work:

- schedulesMode 'holiday' and 'hold' are having no effect
- I can't overwrite the Temperature setting if scheduler is on AUTO
Note: I guess 'Away' from your description above maps to 'holiday', right?
Anything I am doing wrong here?
Another question II have is: Is there a way to switch the display off but keep the thermostat working?
deviveOn:false seems to shutdown the MCU and disables the thermostat functionality.

AlbertWeterings, 2022-01-27:
@Rainerlan Are you sure you have a ME108H? This software is made for the ME102H. As for as I can see there is no support for the ME108H in the originally build @klausahrenberg made. Can you send me pictures of both sides of the PCB that is inside?

I can build you a version for the ME108H but it might show unexpected behavior as it most likely is different. I do not have one so it is not possible for me to dive into the way of communication in between MCU and WiFi module.

Rainerlan, 2022-01-27:
Sorry for the confusion. My Thermostat ist an Avatto ‚WT102‘ according AliExpress shopping cart - so it is a ME102H.
Also the fork works very well, except for the hold and holiday schedules.
What would be really great is a possibility to switch the LCD backlight off, as this is a little bright at night in the bedroom😄 (and I thought the stock FW did not switch off thermostat when long pressing the power key - but yet I have not tried to reflash the stock FW, as all my devices are in use and the devices are now sold at almost twice the price🤑)
Also I have noticed only once, that the WiFi connection was lost while the device was still indicated ,connected‘ (Wi-Fi indicator was not blinking). After disconecting and reconnecting to WiFi on the device UI, the connection came up again.

AlbertWeterings, 2022-01-27:
@Rainerlan I was not able to find more commands the mcu is responding to then now included in the fork. So setting hold and holiday schedules is not possible. Maybe with sniffing communication between WiFi module and mcu it is possible to find more. I broke my original module so am not able to test with the original app.

if I'm right it is possible to set the display backlight in two levels one for day and the other for night. but this is only possible on the thermostat it's self.

If the original app can control the display, hold, holiday schedule I can easily add these functions but first we need to know the right commands.

Rainerlan, 2022-01-27:
Then I will re-flash the stock FW on one of the devices and test the functionality…how can I support to get the traces of the actual MCU commands sent?

AlbertWeterings, 2022-01-28:
When you flash stock firmware first things to do is check if you have control over Display, Hold or Holiday schedule. If not you are ready. If you have then you need to power the PCB with a separate 12 and 5 volt supply (do not power it from mains as you will destroy your computer) Then with a [TTL to USB converter](https://www.banggood.com/Geekcreit-FT232RL-FTDI-USB-To-TTL-Serial-Converter-Adapter-Module-Geekcreit-for-Arduino-products-that-work-with-official-Arduino-boards-p-917226.html?rmmds=product_topselling&cur_warehouse=CN&act_poa=SKU116803) you can start listening on the RX and TX pin of the WiFi module one by one. Use terminal software like [Realterm](https://realterm.sourceforge.io/) This can give you output in HEX. What we need is the commands starting with 55 AA that are send from the wifi module when changing specific functions. Easiest way is start logging to a file and change that function a few times. Also do the same while listening on the RX and TX pin. Then share your log files.