klausahrenberg/WThermostatBeca #248: 1.24beta not working with AVATTO ME102H

Rainerlan, 2022-01-04:
Has the FW version 1.24 beta been tested with Avatto ME102H?
It seems not to be working at all (see problem description below)
The best and most stable version so far for my Thermostat is 1.20n.
1.19 stable does not connect reliably to the MCU - connection indicator keeps blinking
Problem with 1.24 beta on ME102H.
Maybe the Thermostat configuration list is broken and assumes wrong internal settings?
I.e. the list on http://<IP>/thermostat
When I configure the device to ME102H, the propoerties read back as follows:
`http://<IP>/things/thermostat/properties`
the device returns
`{"idx":"thermostat_56930","ip":"<IP>","firmware":"1.24beta","temperature":0.00,"targetTemperature":0.00,"deviceOn":true,"schedulesMode":""}`
I.e. Neither the temperatures are read back correctly, not is the scheduleModes reported correctly.
Also I cannot control ANY properties via the known MQTT topics, such as:
`<DEVICE_MQTT_TOPIC>/thermostat/set/deviceOn false`

AlbertWeterings, 2022-01-04:
@Rainerlan Can you tell us what WIFI module your device has the WBR3 or TYWE3S ?

PhilipTS, 2022-01-07:
@Rainerlan 
If you have a new/latest ME102H have a look at the below issue

https://github.com/klausahrenberg/WThermostatBeca/issues/246

If you swapped out the module (from WBR3 to ESP based) you will need an extra wire between GPIO-2 and GPIO-14

AlbertWeterings, 2022-01-07:
@Rainerlan it looks like you have no communication with MCU Can you check the soldering of the ESP module And more specific the RX and TX pads. You might have damaged the PCB while removing the WBR3 module. In that case don't worry it is easy to repair.

Rainerlan, 2022-01-07:
Sorry, I have not checked the Wi-Fi module yet. But it’s definitively the stock module. Also why does it work fine with firmware version 1.20n but not with 1.24 beta?

AlbertWeterings, 2022-01-07:
@Rainerlan Sorry I get a bit confused. Do you use the original WBR3 module and did you mange to get the firmware loaded on it? If you managed to flash the WBR3 module you are a genius. Can you have a look at the module what is printed on it near the antenna next to the metal cover?

PhilipTS, 2022-01-07:
I will assume that @Rainerlan has an older ME102H that had by default an ESP module (TYWE3S).

AlbertWeterings, 2022-01-07:
@Rainerlan and @PhilipTS if you have the old version with TYWE3S You need to clear the flash chip before flashing 1.24 beta. Then it will work. If you then have problems setting for example target temperature you can try to flash the version that I worked on together with @PhilipTS. I expect it will work without the need of the extra wire. You can find it [here](https://github.com/AlbertWeterings/WThermostatBeca/releases/download/v1.0.6-Beta/WThermostat_1.0.6_Beta.bin) Be aware if you come from 1.19 you always have to erase the flash chip first.

Rainerlan, 2022-01-07:
I checked and can confirm that my device has a TYWE3S module built in. 
@AlbertWeterings : Thanks for the hint with clearing the flash - but I'd rather not open all my 6 thermostats just to clear the flash:)
Usually I do OTA updates only.
As the change log of 1.24 beta is a little spartanic - What is the reason you recommend to change from 1.20n to 1.24 beta?
(I am kind of weighing effort vs. necessity to open and solder all devices again)

AlbertWeterings, 2022-01-07:
@Rainerlan As I see it many small bugs are fixed in between 1.19 1.24B. One of the most important is reconnecting to the network that is fixed. For the last couple of day's I worked with @PhilipTS in a fork of 1.24B to get additional information in the status MQTT message. We now also made changing some settings over MQTT possible.

What we did for example is adding:
MinHeaterTemperature and MaxHeaterTemperature. I use these values to have control over the water temperature flowing trough the radiators. For example if I need to bridge a large temperature difference I steer the heater to a higher temperature to get rooms heated faster the Max setting is used to limit the maximum water temperature in a way there is a good balance in speed of heating without overshooting the target temperature. The Min value is used to control the water temperature as low as possible but still maintain the target temperature. 

To answer your question on the need of doing a update, I always say if you are satisfied with the functionality and stability of the device you use do not change anything ;-)

Rainerlan, 2022-01-17:
After having wired, erased flash and reprogrammed 2 devices, I can confirm that 1.24beta is working almost fine - So the issue may be closed.
Although I have 2 other bugs to report - but probably its better to create a new issue, right?
1) The device ID cannot be changed - the text field in the configuration page tells "max 16 characters" - although all default name is already 18 characters long ("Thermostat_<EspChipID>", where EspChipID is 7 bytes long. I.e len("Thermostat_1234567") = 11+7 = 18 bytes.
Ideally the Device allow would allow the (as I understood) 23 UTF-8 encoded bytes as seems to be stated in MQTT spec.
2) When sending MQTT command to the device, sometimes the commands are only being received by the device at the 2nd attempt (eg. when setting schduledMode on/off.). At the 2nd try, the command is usually received ok.

PhilipTS, 2022-01-17:
@Rainerlan I will assume that you are using klausahrenberg firmware?
If so you should try @AlbertWeterings fork/version as it fixes the second issue.

https://github.com/AlbertWeterings/WThermostatBeca

Rainerlan, 2022-01-18:
@PhilipTS Absolutely! (I am/I was using klausahrenberg firmware)
Now I compiled & tested it. It works awesome! Many thanks for the hint.
The only very tiny comment is, that the TargetTemperature and Temperature topics do now start with uppercase "T":)
Many Thanks for pointing me to the fork!

closed 2022-01-25 by Rainerlan