klausahrenberg/WThermostatBeca #253: deviceOn switching the heating off when commanding deviceOn:false

Rainerlan, 2022-01-25:
As per @PhilipTS, I have flashed the fork from AlbertWeterings, https://github.com/AlbertWeterings/WThermostatBeca on my ME102H devices (Version v1.0.8beta)
I found that it switches the relais/heating off, when I command a deviceOn:false.
Using the "on/off" key on the device has the same effect.

I thought to remember, that the firmware from klausahrenberg really only switched the display off - which is what I'd like to achieve, as the display is pretty bright in the middle of the night.

Is there a way to just switch the display off?

PhilipTS, 2022-01-25:
when you send a deviceOn = false you are turning off the thermostat as if you were pressing the on/off button.
The ESP module send a command to the MCU (DpId1 = 0/1) and the MCU turns off everything.

Rainerlan, 2022-01-25:
Thanks @PhilipTS 
Do you know a way to switch just the display off?
Have you tried the "hold" and "holiday" scheduleModes? I can't figure out how to actually use them.