klausahrenberg/WThermostatBeca #257: Where did MQTT auto-discovery go for HA?

titiviking, 2022-03-04:
Since version 1.19, the MQTT autodiscovery disappeared. Hence the thermostat is not recognized anymore by Home Assistant.

Can this be re-introduced ASAP please ?

neergaby195, 2022-03-09:
You can use this in configurations
############################-----LIVINGROOM -------#######################################
  - platform: mqtt
    name: suf_temp_et
    #availability_topic: "th_suf_et/thermostat/properties/deviceOn"
    #payload_available: "true"
    #payload_not_available: "false"
    
    temperature_command_topic: "th_suf_et/thermostat/set/targetTemperature"
    temperature_state_topic: "th_suf_et/thermostat/properties/targetTemperature"
    current_temperature_topic: "th_suf_et/thermostat/properties/temperature"
    
    mode_command_topic: "th_suf_et/thermostat/set/systemMode"
    mode_state_topic: "th_suf_et/thermostat/properties/systemMode"

    hold_command_topic: "th_suf_et/thermostat/set/systemMode"
    hold_state_topic: "th_suf_et/thermostat/properties/systemMode"
    
    hold_modes: [ "auto", "manual" ]
    payload_on: true
    payload_off: false
    modes: [ "heat", "cool" ]
    min_temp: 10
    max_temp: 32
    temp_step: 1
    precision: 1

titiviking, 2022-03-10:
Thanks for the specs, this could indeed work but I wouldn't exactly call it "auto-discovery" 😅

neergaby195, 2022-03-10:
yep, i know, but it's difficult to make a good configuration for home assistant

titiviking, 2022-03-10:
Native integration into Tasmota? ;-)

titiviking, 2022-03-20:
I've treid the above MQTT config, but I'm unable to make it work properly on my BHT-6000...
There is no such topic as SystemMode, state is not being retrieved properly, HVAC modes are incorrect...

How can I properly debug the whole thing to get some accurate configuration ?

codo70, 2022-11-21:
Tryed it with a ME88H and it also works, only pitty is the missing HA integration..
Thanks for your work so far