klausahrenberg/WThermostatBeca #263: "set" command in manual mode only until next schedule change

fraintt, 2022-05-01:
Hello, I do not know itf this is an issue or not, or how exactly it should work, and maybe somebody could give me an answer, because I was not able to find anything here in the docs. 

When I set the mode to manual, and afterwards also set a temperature, I expect it to last basically forever, until I manually change the mode to auto (to follow the schedule) or after I change the temperature. Now, if I go for ex. on vacation, I want the temperature to be 19 degrees C. This only "lasts" until the next scheduled time. 

If I set it to manual-mode, and also to an temperature, my thermostate changes back to "auto" mode whenever the next scheduled temperature change takes place. 

I´m currently using version: 1.20n

Does anybody else have this same issue, or is it only me? Or is this how the "manual" mode should work?

Bettman66, 2022-05-02:
My thermostats always look on manual and always stay on manual.

fraintt, 2022-05-12:
Hello, 
I´m closing this as it was my error. 
I have set "Auto mode from manual mode at next schedule period change (not at model ET-81W and ME81AH)"
https://imgur.com/e5s4kyX
I have ME81AH, so interresting that it worked! 
Anyway, after unticking this option, and also deleting all retained messages from the mqtt broker, the thermostat did not change the settings.

closed 2022-05-12 by fraintt