klausahrenberg/WThermostatBeca #264: ME81h DP 0x2d & 0x24

danilkorotkov, 2022-05-24:
add DP tracking:
// RX: 55 aa 03 07 00 05 24 04 00 01 **01** crc38
00 - relay ON
01 - relay OFF

// RX: 55 aa 03 07 00 05 0x2d 0x05 0x00 0x01 **0x00** crc41
00 - ok
01 internal sensor broken
02 external sensor broken

c0ldtech, 2022-08-05:
Sorry for hijacking this, but does the ME81H work already? I only found something regarding the ME81AH.

ikke-t, 2022-08-29:
Seems  to be related to #179 which has been open for long. I just got the device, and was wondering how to switch between the sensors. I believe all such int/ext sensor settings are not implemented.

ikke-t, 2022-08-29:
Sorry I was wrong. It seems [there is some code](https://github.com/klausahrenberg/WThermostatBeca/blob/80f9c31756c179941518b3af5650904ce40e52f7/src/WThermostat_ME81H.h#L171) for changing the sensors, just haven't figured yet out how 😀