klausahrenberg/WThermostatBeca #269: Is TYWE3S replaceable by esp8266 in tuya ME81H?

ikke-t, 2022-08-19:
Hi,

I got fooled while ordering ME81H from ebay. They shipped me with no-wifi version. It looks like all components but tywe3s are on board. I was wondering if I could just solder esp8266 from some developer board that I have in place of missing tywe3s and flash this fir,ware?

Is there any change it could work? For the existing pins they lpook compatible, esp has few more.

https://tasmota.github.io/docs/devices/TYWE3S/
https://randomnerdtutorials.com/esp8266-pinout-reference-gpios/

Any experience or knowleqge?

ikke-t, 2022-08-29:
Updating this for the next one:

So I received no-wifi version:
[image]

And I flashed this FW to Wemos D1 mini, and soldered off the ESP12 and soldered it to this board:
[image]

The latter image also has the relay state mod with voltage divider.

And it seems to work fine! Great. Now to study the features of this FW, thanks for all the work on it. Seems to work fine so far.

ikke-t, 2022-08-31:
Just soldered another one. Works fine too. Great :D

closed 2022-08-31 by ikke-t

g6094199, 2023-12-10:
Hi
i also have a rev 1.4 and would like to do the relay mod. can you explain a bit more whats under the tape. is there a divider or just a single R? what size?

ikke-t, 2023-12-10:
It's just a resistor. I don't remember the size. There is another issue somewhere about relay state reading, I likely copied the setup from there.

ikke-t, 2023-12-10:
Hei @g6094199 now that I look it better, I remember. It's divider. See how I put the oyher resistor directly to esp32 pin at the other end of wire.

g6094199, 2023-12-10:
hmm...but the devider must be pulled to gnd somewhere.  maybe unter the shrink somewhere?
anyway. ill put in  a divider  tomorrow.

ikke-t, 2023-12-10:
Yes, the other resistor from the end of wire goes to groud.

g6094199, 2023-12-10:
ahhhhhh....now i see it! ;-)