klausahrenberg/WThermostatBeca #280: Can't change targetTemperature via mqtt

dwarning, 2022-11-10:
I try to set the parameter of my ME81AH over mqtt on my pi, where the broker is running:
`pi@raspberrypi:~ $ mosquitto_pub -d -t Thermostat_Bad/thermostat/properties/targetTemperature -m "23.00"
Client mosqpub|2420-raspberryp sending CONNECT
Client mosqpub|2420-raspberryp received CONNACK (0)
Client mosqpub|2420-raspberryp sending PUBLISH (d0, q0, r0, m1, 'Thermostat_Bad/thermostat/properties/targetTemperature', ... (5 bytes))
Client mosqpub|2420-raspberryp sending DISCONNECT
`
I made also variations for the topic path like 'Thermostat_Bad/cmnd/things/thermostat/properties' or similar.
In the end I have success to change the parameter.

This is my constellation:
`pi@raspberrypi:~ $ mosquitto_sub -v -t '#'
tThermostat_Bad/clock/properties {"ntpServer":"pool.ntp.org","timeZoneServer":"http://worldtimeapi.org/api/ip","epochTime":1668113451,"epochTimeFormatted":"2022-11-10 20:50:51","validTime":true,"timezone":"Europe/Berlin"}
Thermostat_Bad/thermostat/properties {"idx":"Thermostat_Bad","ip":"abc.111.jjj.33","alive":false}
Thermostat_Bad/thermostat/properties/temperature 21.50
Thermostat_Bad/thermostat/properties/targetTemperature 22.00
Thermostat_Bad/thermostat/properties/deviceOn true
Thermostat_Bad/thermostat/properties/schedulesMode off
Thermostat_Bad/thermostat/properties/locked false
Thermostat_Bad/thermostat/properties/systemMode heat
Thermostat_Bad/thermostat/properties/sensorSelection internal
Thermostat_Bad/clock/properties {"ntpServer":"pool.ntp.org","timeZoneServer":"http://worldtimeapi.org/api/ip","epochTime":1668113751,"epochTimeFormatted":"2022-11-10 20:55:51","validTime":true,"timezone":"Europe/Berlin"}
`

PhilipTS, 2022-11-12:
You are publishing your value in the properties.
This is wrong, you subscribe to the properties to receive them but you use a different subject to publish your changes. 
Usually we replace the properties with set but I don;t remember exact details.
Look around and just keep in mind that we subscribe to the properties to receive updates.
We publish on a different subject to make changes.

dwarning, 2022-11-12:
Ok, the world is so simple. Thank's from a mqtt beginner. Replace the subject _properties_ with _set_ works very well.
I struggled because I couldn't see a set entry in JSON tree if I call the IP of the device. Also MQTT explorer shows only properties entries.

dwarning, 2022-11-12:
With the @AlbertWetering fork version 1.0.8_Beta I was able to set TargetTemperature on the ME81AH.

Now with the actual version 1.25_beta from @klausahrenberg the same set command to targetTemperature didn't show effect, whether to display nor to MQTT Explorer.

Question is: How and by which tool can I see the list of "set"able variables? Is there web page for the IP of this device? If I put in the IP alone I see a JSON tree only with properties.

AlbertWeterings, 2022-11-19:
@dwarning 
> Question is: How and by which tool can I see the list of "set"able variables? Is there web page for the IP of this device? If I put in the IP alone I see a JSON tree only with properties.

When I started working on my ME102H device I have used a NodeMCU to capture the communication in between the WiFi/Tuya module with the original firmware and App and the thermostats MCU. This was working reasonable well. At some point I started to work with @PhilipTS and switched to using the program [Realterm](https://realterm.sourceforge.io/). **Be aware you can't connect a serial interface to the thermostat when you power it from mains!! this will blow up your equipment!! and is dangerous**

Then while monitoring the data changing all settings in the App it was easy to capture all possible settable variables. Also the Tuya Cloud Developer Platform revealed some of the functions. This helped @PhilipTS and me to find all settable variables for the ME102H thermostat.

So if you want to know all the supported commands this is I think the way to go. I did not see anyone who has documented them all yet.

robertorobles, 2022-11-19:
@dwarning to send the command is not properties, if not set
can you try this?
`mosquitto_pub -d -t Thermostat_Bad/thermostat/set/TargetTemperature -m "22.00"`

dwarning, 2022-11-19:
It is OK with v. 1.0.10 - I can set it up and down.

Am 19.11.22 um 20:21 schrieb Roberto Robles Fernández:
> |mosquitto_pub -d -t Thermostat_Bad/thermostat/set/TargetTemperature 
> -m "22.00"|

closed 2022-11-21 by dwarning