klausahrenberg/WThermostatBeca #282: ME81AH problem with sensorSelection (null)

dwarning, 2022-11-19:
I am using the last firmware from @AlbertWeterings repo version 1.0.8 Beta.

Any time I make a total power up I got tons of following message from my broker:
` mosquitto_sub -v -t Thermostat_Bad/#
Thermostat_Bad/clock/properties {"ntpServer":"pool.ntp.org","timeZoneServer":"http://worldtimeapi.org/api/ip","epochTime":1668845184,"epochTimeFormatted":"2022-11-19 08:06:24","validTime":true,"timezone":"Europe/Berlin"}
Thermostat_Bad/thermostat/properties {"idx":"Thermostat_Bad","ip":"yxc.qwe.ase.pp","alive":false}
Thermostat_Bad/thermostat/properties/deviceOn true
Thermostat_Bad/thermostat/properties/schedulesMode off
Thermostat_Bad/thermostat/properties/locked false
Thermostat_Bad/thermostat/properties/systemMode cool
Thermostat_Bad/thermostat/properties/Temperature 21.50
Thermostat_Bad/thermostat/properties/TargetTemperature 22.00
Thermostat_Bad/thermostat/properties/sensorSelection (null)
Thermostat_Bad/thermostat/properties/sensorSelection (null)
Thermostat_Bad/thermostat/properties/sensorSelection (null)
Thermostat_Bad/thermostat/properties/sensorSelection (null)
Thermostat_Bad/thermostat/properties/sensorSelection (null)
Thermostat_Bad/thermostat/properties/sensorSelection (null)
Thermostat_Bad/thermostat/properties/sensorSelection (null)
.
.
`
If then set the variable by:
`mosquitto_pub -t 'Thermostat_Bad/thermostat/set/sensorSelection' -m "internal"`
everything is OK.

Seems the init of the variable sensorSelection is not working.

AlbertWeterings, 2022-11-19:
@dwarning Can you please check what [this version](https://github.com/klausahrenberg/WThermostatBeca/issues/239#issuecomment-1320848180) is doing. I made a couple of changes that might also fix this issue.

dwarning, 2022-11-19:
I updated to 1.0.10. Made no changes in config. Div. set commands are working. But the empty sensorSelection variable still exists and brings a tornado of the shown messages. The problem is that the fhem logfiles will explode my small SD card.
The cppcheck warning show still the the uninitialized variable as mentioned in #239.

AlbertWeterings, 2022-11-19:
@dwarning also test 1.0.11-beta I found the initialized sensor selection byte was set to 2b. I found for ME81H most likely it should be 2d. Hope this solves the issue

dwarning, 2022-11-19:
Sorry I am not a C++ guy.

You are sure that it is the byte? I assume that the 3 sensor strings 
come not in the init.

Because if I set over mqtt the sensorSelection the message disappears.

Its not easy to start the thermostat because he is installed. I must cut 
him from supply because a simple reboot is not sufficient.

dwarning, 2022-11-19:
I can confirm that the problem is solved by v1.0.11. No sensorSelection (null) anymore after cold start.
Thank you!

closed 2022-11-19 by dwarning