klausahrenberg/WThermostatBeca #284: Unable to Flash Beca BAC-6000

mojtabavi, 2022-12-01:
I'm Trying to flash beca BAC-6000 thermostat using CH340G device
But the desired esp module is not placed in the program mode and sends incomprehensible characters on the serial
ESP Model:
TYWE3S
Connection:
esp-gnd --> gnd
esp-3v3 --> 3v3
esp-gpio0 --> gnd
esp-rx --> tx
esp-tx --> rx
[image]
[image]

I'm really confused, I'd really appreciate it if someone could help

mojtabavi, 2022-12-01:
https://user-images.githubusercontent.com/63472358/205035911-6b0a889b-5282-4a95-8274-f76a5ea50865.mp4

closed 2022-12-04 by mojtabavi

jfjeronimo, 2023-02-12:
could you share the solution ? I have the same problem