klausahrenberg/WThermostatBeca #286: ME81AH problem with WThermostatBeca_weterings V 1.1.0

dwarning, 2022-12-06:
@AlbertWeterings:
I updated from version 1.0.11 beta to latest 1.1.0 beta.
Few issues:

- WiFi symbol missing, but connection to web config was possible
- Info page not accessible, other pages OK
- MQTT connection seems broken - no update

So I downgraded again to 1.0.11 and everything works.

AlbertWeterings, 2022-12-07:
@dwarning

I was able to reproduce on my ME102H only:
- Info page not accessible, other pages OK
The other two problems I did not see on my Thermostat. Created a new release [Beta-1.1.1](https://github.com/AlbertWeterings/WThermostatBeca/releases/tag/v1.1.1-beta) This solved the info page on my thermostat the WiFi symbol was still there and MQTT was updating.

Let me know if this solves your issues to?

dwarning, 2022-12-08:
How you make the Update? 
If I make the OTA update I get lot of trouble with my HA system FHEM. The symbol is away and some times I have to set some settings like the sensor usage. If I make then a cold start by switching power total off/on everything is OK. So I am fearing a bit to change the running system.

AlbertWeterings, 2022-12-09:
@dwarning I update my ME102H via OTH.
I think what is causing this issues is the change of the library WAdapter-1.24 to WAdapter-1.25 and the update of some other ESP8266 libraries. These libraries are updated automatically by platform IO something I have no control over.

AlbertWeterings, 2022-12-09:
@dwarning, on my thermostat I checked if there is a difference in MQTT communication and I see no difference. I went back to 1.0.9 and the latest version.

dwarning, 2023-01-20:
Firmware 1.0.11 works reliable.

closed 2023-01-20 by dwarning