klausahrenberg/WThermostatBeca #290: Where to buy devices with esp chipset?

btk999, 2022-12-31:
Hi guys,

i ordered some devices at aliexpress / amazon, but all of them have incompatible chips and i dont want to solder.
Does somebody know a good shop/source that sells devices compatible to WThermostatBeca with esp8266 chip?

Thanks and regards

Trashmee, 2023-01-02:
Hi @btk999,

not sure your question is accurate - but I might simply missunderstand it..
 - All BHT-002 devices I ordered from Aliexpress have an ESP8266. 
 - Make sure you get Wifi & the right type for your heating system
 - You WILL have to solder once in order to flash the klausahrenberg FW. Once you're there online update via WebIF will be supported. The earlier tuya-convert process is not working any more, I believe that's true for a lot of different devices (e.g. also for the power meter I bought)

Hope this helps,
Alex

Trashmee, 2023-01-02:
Reading this one (https://github.com/fashberg/WThermostatBeca/issues/171) - I might be wrong. Maybe I was lucky with my "BHT-002 GALW" that I received two weeks ago.

btk999, 2023-01-02:
Thanks for your response @Trashmee, but all the BHT-GBLW thermostats i ordered at aliexpress recently did not have the esp8266 chip anymore. i ordered one at moes and one from another shop and they both had the cb3s chipset which is not compatible to this firmware.
So my idea with this issue was to find sellers that ship the bht-002 with the old esp8266 chipset (or any other compatible model of this firmware).
Can you post the url to the seller where you ordered your model?

Trashmee, 2023-01-02:
Mine is from a store on AliExpress called [MoesHouse Direct Store](https://www.aliexpress.com/store/5476242?spm=a2g0o.order_list.order_list_main.7.480e5c5fDvMFmn) - but reading the issue 171 above it might be a model specific change. No clue - good luck!

pavelhrolovich, 2023-01-11:
Well, I have got BAC-002 with CB3S. I can confirm simple re-soldering to ESP8266 in the same form factor works fine.

shakin89, 2023-02-07:
> Well, I have got BAC-002 with CB3S. I can confirm simple re-soldering to ESP8266 in the same form factor works fine.

Which type of esp8266 form factor did you solder on? i have a bunch of bac-002-alw or elw (2 or 4 valves) which i'd like to disconnect from tuya cloud

cparg, 2023-04-22:
I bought a BecaSamrt BHT-002 some weeks ago and it also came with a CB3S.
After several attempt to falsh that Beken 7231 device I gave up. 
I bougth some ModeMCU with ESP 8266 and flashing WThermastatBeca worked so far so good.
I replaced the CB3S by the ESP8266 but the Display shows only a "lock" symbol.

The hardware wiring to the MCU seem to be different. There are now some transistors on the RxX TxD Pins.
Any hint ? This is how it looks like :
[image]

SilverHazeLazers, 2023-05-25:
Been a month or so since the last update, has anyone had any luck finding devices on aliexpress that ship with an esp chip?