klausahrenberg/WThermostatBeca #292: Humidity Sensor Addon?

malgates, 2023-01-23:
Hi everyone,

is it possible to add a humidity sensor to the ESP? Does not need to display it via the display. But sharing the info via MQTT would be great!

andreasz70, 2024-01-06:
I am just thinking about integrating a DHT22 temperature/humidity sensor. There should be enough space on the BHT002 board for the sensor. It could be attached to a free GPIO of the ESP module. The WThermostatBeca-software could altered including the DHT library and handling the signal similar to the GPIO5 relay input (optional switch in configuration, serve value via MQTT or webthings). Looks quite simple but I am not got enough in software to do that fast.

DerDom, 2024-02-08:
I would love to add a humidity sensor to the ESP, as well!

andreasz70, 2024-02-15:
I added a few lines to the source code:
- include DHT.h library, 
- init dht object, 
- add a checkbox to the web-config
- add a few lines to activate the humidity property
and modified the hardware to connect a DHT22 sensor to GPIO 4 (tested without the BHT002 connected)

How to upload the code?
[image]
[image]

malgates, 2024-04-09:
How does the H/W mod look like? Will this be merged into the main branch?

THX

malgates, 2024-04-09:
How dows the Hardware mod look like?

Unfortunately I closed the issue. Is it possible to open it again? Can I test the branch? I'm a github noob :-D

closed 2024-04-09 by malgates