klausahrenberg/WThermostatBeca #296: Build toolchain

cparg, 2023-03-25:
Hello,
the build instructions only mention Arduino or Atom IDE and a broken link to some library dependencies.
Can you provide some more information (or point me to) how to build the firmware ?
Does it build also for a BHT-002 that has a CB3S module with an BK7231N MCU (RISC-V) ?
Many thanks
Chris

cparg, 2023-04-22:
I stepped back to use a plain ESP32 (on a NodeMCU v2).
Still the build process lacks a Timer.h include file. 
Does anybody have a hint where this is supposed to come from ?
```
Compiling .pio/build/nodemcuv2/libf13/ESP8266WiFi/ESP8266WiFiMulti.cpp.o
In file included from src/WThermostat.cpp:3:

src/WClock.h:15:10: fatal error: Time.h: No such file or directory

**************************************************************
* Looking for Time.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Time.h"
* Web  > https://registry.platformio.org/search?q=header:Time.h
*
**************************************************************

   15 | #include <Time.h>
      |          ^~~~~~~~
compilation terminated.
```

cparg, 2023-04-23:
This is also fixed in latest version.

wardhou, 2024-01-14:
Hello, does this code work on a BHT-002 that has a CB3S module with an Beken BK7231N MCU ?
Does anyone has a builded firmware file, that i can flash on the CB3S module ?

g6094199, 2024-01-14:
no

then you have to go to: https://github.com/OpenBekenIOT