klausahrenberg/WThermostatBeca #302: How to Debug?

cparg, 2023-04-22:
Hello,
I am trying to understand how the Firmware can be debugged.

- Which Pins are used to communicate with the MCU ? Where is that configured in the FW?
  - This version of the BHT002 Hardware I have has TxD and RxD pins of the ESP connected to some transistors. 
- How can I obseve the network()->debug() messages I can find in the code ?
  - As TxD is already used by something, is there a second Serial Interface ? Which Pin?
  - The network() indicates that the debug messages shall be also visible though the networks somehow. 
    How ? Do I need to open http://<IPv4>/<something> ?

The Display only shows a little lock symbol and the device does not react on any of the touch keys. I am wondering if the ESP  can properly communicate to the MCU at all. 

Hard to figure this all out by browsing the sources. Some helping hand would be highly appreciated.

Many thanks
Chris

klausahrenberg, 2023-05-17:
Hi,
Tx and Rx can't be configured in this firmware. If Tx and Rx are connected to other pins at your Tuya device, variable pins must be implemented.
network()->debug() messages will be forwarded by mqtt and to serial output, if DEBUG flag is set to true in firmware. On precompiled binaries, this flag is not activated - so no effect.

cparg, 2023-05-18:
Hello Klaus, 
many thanks for your response. 
Which one is the "serial port" !? TxD and RxD are quite generic terms for a serial ports as well but I suppose there is a second TxD that's used as a "debug console" !?
Many thanks
Chris