klausahrenberg/WThermostatBeca #308: Fork Merging?

Stoatwblr, 2023-08-01:
Hi Klaus

There are a number of good ideas in the other forks (eg: console debug logging) which would be great to fold back into your "mainline" in order to avoid excess fragmentation of WthermostatBeca

Is there any chance of considering doing that?

kalwinskidawid, 2023-10-25:
Hey, could you provide an exact link to them?

g6094199, 2023-12-23:
> Hi Klaus
> 
> There are a number of good ideas in the other forks (eg: console debug logging) which would be great to fold back into your "mainline" in order to avoid excess fragmentation of WthermostatBeca
> 
> Is there any chance of considering doing that?

normaly the forker sets an pull request on the changes to the upstream  source. so this might be better communicated to the forkers issues, maybe with a link to here.

but in general getting fresh ideas and code upstream is always a good idea (fragmentaion and long term support wise)!

Stoatwblr, 2024-06-04:
> Hey, could you provide an exact link to them?

I'm thinking of the fas fork, if you're not already checking it