klausahrenberg/WThermostatBeca #310: Homeassistant does not see the thermostat via MQTT.

mistikdonbass, 2023-10-16:
Good afternoon  I can't pair the thermostat with homeassistant via MQTT.  My thermostat and settings in screenshots.
 What did I miss, what did I do wrong?
[image]
[image]
[image]
[image]
[image]
[image]
[image]
I also tried installing another firmware from this topic: https://github.com/fashberg/WThermostatBeca/tree/master 
with this firmware it connects to homeassistant, but does not transmit information about temperatures (set and actual).

k-zakhariy, 2023-10-16:
@mistikdonbass what's your configuration on HA side ?

mistikdonbass, 2023-10-17:
> @mistikdonbass what's your configuration on HA side ?

Thanks for the quick response.

Judging by MQTT Explorer, it connects to HA and transfers data. But it does not appear on devices.
[image]
[image]

Arquiteto, 2023-10-29:
There is no auto discovery on this firmware so you must configure your MQTT climate entity manually. Here is some example code, but you need to adjust to your topics.

```
mqtt:
  climate:
      - name: Łazienka sterownik podłogowy
        temperature_command_topic: "termostat_lazienka/thermostat/set/targetTemperature"
        temperature_state_topic: "termostat_lazienka/thermostat/properties"
        temperature_state_template: "{{ value_json['targetTemperature'] | round(1) }}"
        current_temperature_topic: "termostat_lazienka/thermostat/properties"
        current_temperature_template: "{{value_json['temperature'] | round(1) }}"
        mode_state_topic: "termostat_lazienka/thermostat/properties"
        mode_state_template: >
          {% if value_json.deviceOn != true %}
          off
          {% elif value_json.schedulesMode == 'auto' %}
          auto
          {% else %}
          heat
          {% endif %}
        modes:
          - "heat"
          - "auto"
          - "off"
        min_temp: 10
        max_temp: 30
        temp_step: 0.5
        precision: 0.5
        retain: true
```

mistikdonbass, 2023-10-30:
> There is no auto discovery on this firmware so you must configure your MQTT climate entity manually. Here is some example code, but you need to adjust to your topics.

Thanks a lot.
I already understood this and used a more complex config.
Works too.

```
mqtt: #Настройка термостата для бойлера
  climate:
    - name: Thermostat
      unique_id: termostat
      device:
        configuration_url: http://192.168.1.166/config
        manufacturer: Tuya
        model: ME81AH
        connections: [["mac", "E8: DB: 84:D1: EF:33"]]
      temperature_command_topic: thermostat/thermostat/set
      temperature_command_template: >
        {{ {'targetTemperature':value} | to_json }}
      temperature_state_topic: thermostat/thermostat/properties
      temperature_state_template: >
        {% if value_json is defined and value_json['targetTemperature'] is defined %}
          {{ value_json['targetTemperature'] }}
        {% endif %}  
      current_temperature_topic: thermostat/thermostat/properties
      current_temperature_template: >
        {% if value_json is defined and value_json['temperature'] is defined %}
          {{ value_json['temperature'] }}
        {% endif %}  
      mode_state_topic: thermostat/thermostat/properties
      mode_state_template: >
        {% if value_json is defined and value_json['deviceOn'] is defined %}
          {% set deviceOn = value_json['deviceOn'] %}
        {% else %}
          {% set deviceOn = false %}
        {% endif %}
  
        {% if value_json is defined and value_json['schedulesMode'] is defined %}
          {% set schedulesMode = value_json['schedulesMode'] %}
        {% else %}
          {% set schedulesMode = 'off' %}
        {% endif %}
  
        {% if deviceOn == true %}
          {% if schedulesMode == 'auto' %}
            auto
          {% else %}
            heat
          {% endif %}
        {% else %}
          off
        {% endif %}
      action_topic: thermostat/thermostat/properties
      action_template: >
        {% if value_json is defined and value_json['systemMode'] is defined %}
          {% set deviceOn = value_json['deviceOn'] %}
        {% else %}
          {% set deviceOn = false %}
        {% endif %}
        
        {% if value_json is defined and value_json['systemMode'] is defined %}
          {% set systemMode = value_json['systemMode'] %}
        {% else %}
          {% set systemMode = 'off' %}
        {% endif %}
        
        {% if deviceOn == true %}
          {% if systemMode == 'cool' %}
            idle
          {% elif systemMode == 'heat' %}
            heating
          {% endif %}
        {% else %}
          off
        {% endif %}
      mode_command_topic: thermostat/thermostat/set
      mode_command_template: >
        {% if value == 'auto' %}
          {{ {'deviceOn':true,'schedulesMode':'auto'} | to_json }}
        {% endif %}
  
        {% if value == 'heat' %}
          {{ {'deviceOn':true,'schedulesMode':'off'} | to_json }}
        {% endif %}
  
        {% if value == 'off' %}
          {{ {'deviceOn':false,'schedulesMode':'off'} | to_json }}
        {% endif %}
      modes:
        - "auto"
        - "heat"
        - "off"
      temperature_unit: C
      temp_step: 1
      min_temp: 5
      max_temp: 85
      initial: 20
      retain: true
      qos: 0

```

Your proposed configuration makes more sense to me. But I don’t understand why the thermostat doesn’t turn off.
If you click on turn off.

Arquiteto, 2023-11-01:
The template you posted works better. I had a very basic one for a different thermostat which was only to serve as an example. This works better (I have installed ME81H just today so I was able to check)

yhunter-ru, 2025-02-12:
My HA template for BHT-002:

```
mqtt:
  climate:
    - name: Thermostat
      unique_id: termostat
      device:
        configuration_url: http://192.168.1.187/config
        manufacturer: Moes
        model: BHT-002
        connections: [["mac", "98:F4:AB:BF:78:D1"]]
        
      current_temperature_topic: thermostat_12548305/thermostat/properties/temperature
      temperature_command_topic: thermostat_12548305/thermostat/set/targetTemperature
      temperature_state_topic: thermostat_12548305/thermostat/properties/targetTemperature
      mode_state_topic: thermostat_12548305/thermostat/properties/schedulesMode
      mode_state_template: >
        {% if value == 'auto' %}
          auto
        {% else %}
          heat
        {% endif %}

      mode_command_topic: thermostat_12548305/thermostat/set/schedulesMode
      mode_command_template: >
        {% if value == 'auto' %}
          auto

        {% else %}
          off
        {% endif %}

      modes:
        - "auto"
        - "heat"
      temperature_unit: C
      temp_step: .5
      min_temp: 15
      max_temp: 35
      initial: 25
      retain: true
      qos: 0
```