klausahrenberg/WThermostatBeca #318: Support for Avatto WT410-BH

ingerul93, 2024-01-12:
Hello,

I was wondering if you can add support for this device:
https://vi.aliexpress.com/i/1005006109579911.html

I have the device and I can be a tester for it but I do not know from where to start.

Best regards,
Ioan Vlad

ingerul93, 2024-01-12:
[image]

ingerul93, 2024-01-12:
The chip from the left is CA51F253L3-S2  CACHIP. 

Basic features
 Core
 CPU: 1T 8051, the maximum speed is 10 times faster than the traditional 8051
 Compatible with 8051 instruction set, dual DPTR working mode
 CPU frequency: up to 24MHz supported
 Memory
 Flash: 8 / 16 / 32K bytes, supported Repeatedly erase and write multiple times
 Flash can be divided into program space and data space. The data space can be used to store data that needs to be saved when power fails. EEPROM can be omitted
 RAM: 256 bytes internal RAM, 2K bytes external RAM
 Working voltage
 Working voltage : 1.8 - 5.5V wide voltage operating range
 Clock system
 External high-speed oscillator: 1 - 24MHz
 External RTC oscillator: 32.768KHz
 Built-in low-speed RC oscillator: 131KHz
 Built-in PLL: frequency multiplication multiple is 2 - 10 times, Reference clock is 2 - 4MHz built-in RC oscillator
 Built-in high-speed RC oscillator: 2 - 4MHz, accuracy up to 1% (factory initial frequency is 3.6864MHz＠3.3V/25℃)
 Built-in external clock monitoring module, can effectively Monitor the working status of various external clocks to avoid crashes caused by external clock oscillation
 C RTC function
 The built-in RTC module can time, minutes, seconds, days of the week, and days, and supports the alarm clock function
 Supports millisecond and half-second interrupts
 Interrupt system
 15 Effective interrupt sources
 Two levels of interrupt priority, supporting interrupt nesting
 10 external interrupt sources, each external interrupt can be configured with any signal pin as an interrupt input pin
 Timer
 Three 16-bit general-purpose timers: timer 0, timer 1, timer 2
 General-purpose input and output port (GPIO)
 Supports up to 62 GPIO ports, supporting push-pull, open-drain, pull-up, and pull-down , High-impedance mode
 When pushing-pull output, a single GPIO push current supports 20mA, and sink current supports 40mA
 Touch Key (Touch Key)
 Built-in touch sensor controller
 Supports a maximum of 24 touch channels
 High anti-interference performance, compliant with EMC (CS) ) Standard
 Can work normally in STOP mode, supports touch wake-up in STOP mode
 Analog//digital converter (ADC)
 Supports 8-channel 12-bit SAR ADC, built-in op amp and comparison function
 Supports 3 reference voltage sources: VDD , internal reference, external reference
 The VDD voltage can be measured when the internal voltage is selected as the reference voltage
 Supports detection signal reduction and amplification functions, the scaling factor is optional
 PWM
 Supports 8-channel PWM, and the period and occupancy can be configured arbitrarily within the 16-bit range Duty ratio
 Support complementary mode and dead zone control, can be used to drive brushless DC motors
 Support settable edge alignment and center alignment modes
 Support direct output of internal clock function
 Support PWM interrupt
 LCD driver
 Support up to 8com x 32seg , 7com x 33seg, 6com x 34seg, 5com x 35seg, 4com x 36seg
 Configurable duty cycle: 1/2, 1/3, 1/4, 1/5, 1/6, 1/7, 1/8 Duty
 Configurable bias voltage: 1/2, 1/3, 1/4 Bias
 Support 8 levels of contrast adjustment
 Support 3 levels of drive current, users can adjust according to different LCD screens
 LED driver
 Support up to 8com x 32seg
 Supports 8 levels of brightness adjustment
 Low voltage detection (LVD)
 Configurable voltage detection range 1.8 - 4.8V
 Can set low voltage reset or interrupt
 Reset mode
 The chip supports multiple reset sources: hard reset, soft reset, Watchdog reset, low voltage detection reset, power-on/power-down reset
 Watchdog
 27-bit watchdog timer, 16-bit adjustment accuracy, configurable watchdog reset or interrupt

 Remote control receiving function
 Built-in sampling counter module (SAMPLE), which can sample pulse widths of any length through hardware modules, reducing software code
 Universal serial interface (UART)
 Supports up to 3 UART interfaces
 Supports 1-byte receive buffer
 I SPI interface
 Built-in 1 4-wire SPI interface, supports master-slave mode
 II 2 2 CC interface
 Built-in 1 I 2 C interface, supports master-slave mode, supports standard/fast/high-speed mode
 Operational amplifier and analog comparator
 Supports 4 analog comparators, two operational amplifiers and a capture counter
 The analog comparator reference voltage can select an internal reference or an external input reference
 The analog comparator has a built-in 15-bit digital filter and supports comparator interrupts
 The operational amplifier can be combined with ADC and analog comparator are used together to expand the detection signal voltage range
. Capture counter and analog comparator are used together to measure motor speed and stall detection.
Brushless DC motor drive
. Built-in 60° Hall and 120° Hall decoding. Module
 Supports automatic mode and manual mode, supports braking function
 Supports multiple abnormality detection
 Combined with analog comparator, it can realize Huo-free DC motor drive
 Multiplier and Divider (MDU)

CraigHutchinson, 2024-02-18:
I just got one of these hoping can put tasmota on it.... WBR3 can be replaced with ESPC2-16 as far as I know but not looked any further yet.

Which chip is the one on the left you mention?

ingerul93, 2024-02-18:
This is the chip 
[image]

ingerul93, 2024-02-18:
Here is a list of commands that I have found.
[cmds.txt](https://github.com/klausahrenberg/WThermostatBeca/files/14324406/cmds.txt)

CraigHutchinson, 2024-02-18:
This should help https://developer.tuya.com/en/docs/iot/tuyacloudlowpoweruniversalserialaccessprotocol?id=K95afs9h4tjjh

CraigHutchinson, 2024-02-18:
This guys work may be a helpful resource for Tasmota and may be worth checking albeit for an AC controller https://github.com/arendst/Tasmota/discussions/19236

ingerul93, 2024-02-20:
I plan to do the same thing with WBR3 chip. I already checked the docs and I can't find anything related to : 55 AA 3 7 0 5 B 5 0 1 0 which for me is an unknown command.

g6094199, 2024-02-20:
why not giving [openBk](https://github.com/openshwprojects/OpenBK7231T_App) a try, which behaves almost like tasmota

CraigHutchinson, 2024-02-20:
> why not giving [openBk](https://github.com/openshwprojects/OpenBK7231T_App) a try with the WBR3, which behaves almost like tasmota

It could be nice to improve that projects description... So does that build a firmware that can go onto the WBR3 itself?

Asside from this I checked the pinout and looks very minimal (WBR3 image from somebody else) 
[image]

g6094199, 2024-02-20:
agree.....the project docu is more than "improvable". yes, you flash an os to the tuya mcu. the youtube videos are more self explaining...

ingerul93, 2024-02-20:
> why not giving [openBk](https://github.com/openshwprojects/OpenBK7231T_App) a try with the WBR3, which behaves almost like tasmota

Given that I have a custom configuration(hardware and soft) I want to use something created by me and not some third party implementation.

CraigHutchinson, 2024-02-20:
I foudn this bit of niceness today... another is on order! ESP32-C6 (Latest version) but in a ESP12F compatible package 😎https://www.aliexpress.com/item/1005006435652251.html

CraigHutchinson, 2024-02-21:
Of interest, I checked https://github.com/openshwprojects/OpenBK7231T_App and the WBR3 isn't going to be suppoerted soon I expect. Its a newer generation of Tuya MCUs rocking an KM4 which is a completely new architecture from  the `BK*` compatible devices. There are some thrreads on the project, but the Tuya website they list the `WB Series` and `WBR Series` as separate sections for this.

CraigHutchinson, 2024-02-22:
So I captured some serial comms just now. 

I'm not sure if this relates to the other fault message noted on here but I noticed entering pairing mode didn't work while capturing the comms. 

This is due to the TX from the WBR3 where the other mcu wasn't seeing high enough signal levels to receive the messages sent back. 

There is a Diode on the TX so this would drop 0.5-1.0v plus me snooping on the signal it makes sense. Not sure if it will be okay to bypass that diode or not yet.

kelogs, 2024-12-11:
Hi,

Does anyone know where to find a datasheet for the CA51F253L3 MCU ? Or what the original MCU / a compatible MCU would be ?

thanks

MrdUkk, 2025-01-09:
@kelogs  yes. sure.  
for example here is PDF CA51F2x series chips https://file.elecfans.com/web1/M00/B7/93/o4YBAF5vG56AXth_ADmOjmuWO5M891.pdf

for discovering and learning tuya RF module communication with central MCU i wrote simple python shiffer. https://github.com/MrdUkk/tuya_sniffer