klausahrenberg/WThermostatBeca #329: Avatto ME81H - new model with WBR3?

os63, 2024-12-20:
I've got a set of Avatto ME81H thermostats, that came with TYWE3s chips - which I replaced with ESP-12F pre-flashed to WThermostatBeca. These have worked well.

However, the new set of Avatto ME81H thermostats that just came, are with WBR3 chips. The trickery of replacement with ESP-12F now yields a thermostat with Wi-Fi icon off (and not a peep on WLAN).

Anyone else with the same problem, and any solutions available?

Thank you.

g6094199, 2024-12-20:
i guess the enable or other PIN is not connected (correctly) on these versions. no traces/resistors?  check this.

os63, 2024-12-20:
Enable lifted off the thermostat and hardwired to VCC; and behold, we are live. Not with the Wi-Fi icon, but don't care.

Thank you! Saved my week.

Langer75, 2025-01-21:
I run in the same pronblem. What do you mean with "lifted off"? Can you described it a little bit more please? Thanks in advance

os63, 2025-01-22:
> I run in the same pronblem. What do you mean with "lifted off"? Can you described it a little bit more please? Thanks in advance

Ah.. This means, that when soldering in a standard espressif chip, use a temporary spacer or similar to have the replacement chip so, that none of the contacts touch the thermostat board. This way you can control, which pins actually are connected and which are not (via solder bridges).

Langer75, 2025-01-23:
is this only for flashing?Therfor I use a pogo adapter. Otherwise you have to use this temporary spacer at every powercycle. Or do I understand that wrong? I think only EN, VCC, GND, GPIO15, RX and TX are connected

os63, 2025-01-23:
> Is this only for flashing? Therfore I use a pogo adapter. Otherwise you have to use this temporary spacer at every powercycle. Or do I understand that wrong? I think only EN, VCC, GND, GPIO15, RX and TX are connected

This is the to-be permanent installation.

On newer ME81Hs originally with WBR3, the Espressif chip won't start otherwise.

Langer75, 2025-01-23:
And which pins needs to be connected? I think at least RX and TX have to be connected, beside VCC and GND.

Langer75, 2025-01-28:
OK now I got it the ENABLE pin needs to be lifted of and hardwired to VCC. Sorry was my bad understanding

os63, 2025-01-29:
Good, good. Sorry about that - I did a couple re-routes on mine, but did not write down what. It'll probably change again anyway for the next batch they manufacture.
Did you get the WLAN icon on the thermostat display? I did not.

Langer75, 2025-01-29:
I also did not get an WLAN icon. But it doesn't matter, as all other functions work as expected.